gbionics/human-dynamics-estimation #152: Remove base position offset in human-dynamic-estimation

lrapetti, 2019-10-08:
The idea is to add rpc functionality to the human-state-provider in order to be able to remove an offset for the base position. In this way it would be possible to take care of the relative transform between two humans.

Given some deadline requirements, this feature will be first implemented on top of the following feature branch: [`feature/visualize-berdy-estimated-wrench`](https://github.com/robotology/human-dynamics-estimation/tree/feature/visualize-berdy-estimated-wrench)

lrapetti, 2019-10-08:
CC @prashanthr05 @Yeshasvitvs @kouroshD

lrapetti, 2019-10-08:
It has been implemented in https://github.com/lrapetti/human-dynamics-estimation/commit/08b75c27b718e2ab632b482146188a26ff36ec28.

It allows to remove the position offset for one or more axis, and the rotation offset for one or more axis. Preliminary test have been completed, further testing will follow,

lrapetti, 2019-10-11:
PR opened to the feature branch (https://github.com/robotology/human-dynamics-estimation/pull/153)

claudia-lat, 2019-12-10:
In the meanwhile I'm moving this issue onto the Iteration.

lrapetti, 2020-01-27:
PR has been merged, closing.

closed 2020-01-27 by lrapetti