gbionics/human-dynamics-estimation #154: Write an yarpmanager application file for bash script for retargeting visualization

kouroshD, 2019-11-05:
According to:
https://github.com/dic-iit/component_wholebody-teleoperation/issues/194

- It happens that Xsens calibration process, returns good as the result of the calibration, but in fact it is not good. For this reason we need to check the results using visulaization.

  - We Will check if we can run all the visualization parts, in a yarpmanager application file, since we have a roslaunch for rviz. Otherwise, we will write a bash file for that to increase the speed for launching the visualization, similar to AnDy second review meeting using shell scripts or yarp manager application https://github.com/dic-iit/retargeting-robot-models/tree/master/retargeting-shell-scripts.

kouroshD, 2019-11-11:
is done and tested with this commit: [HDE#187f822](https://github.com/kouroshD/human-dynamics-estimation/commit/187f8229fb90c4ba8884fa757dbf7e2b59e4a895)

kouroshD, 2019-11-11:
The PR link: https://github.com/robotology/human-dynamics-estimation/pull/155

lrapetti, 2019-11-16:
PR merged. Closing

closed 2019-11-16 by lrapetti