gbionics/human-dynamics-estimation #159: Update rpc port reader class handling in various devices

Yeshasvitvs, 2019-11-20:
The two main devices of HDE that has rpc functionality are:
-  `HumanStateProvider` https://github.com/robotology/human-dynamics-estimation/blob/feature/visualize-berdy-estimated-wrench/devices/HumanStateProvider/HumanStateProvider.cpp#L155 
- `HumanDynamicsEstimator` https://github.com/robotology/human-dynamics-estimation/blob/feature/visualize-berdy-estimated-wrench/devices/HumanDynamicsEstimator/HumanDynamicsEstimator.cpp#L968. 

The class name for providing rpc functionality are named the same as `CmdParser` in both the devices without any namespace. When launching `HDE` on a linux machine, I notice a strange behavior of the `HumanDynamicsEstimaor` device rpc port. Using `help` command on this rpc port, I was seeing the available options from the help of `HumanStateProvider` rpc server. This problem does not seem to exist on MacOS as @lrapetti was running the same code on his machine.

Thanks to @diegoferigo the problem seems to be something he faced before with shared libraries https://github.com/robotology/gym-ignition/issues/101. 

As both `HumanStateProvider` and `HumanDynamicEstimator` are shared libraries (plugins) that are run in a single `yarprobotinterface` with `HumanStateProvider` first and `HumanDynamicEstimator` after, the use of same class name as `CmdParser` in both the devices seems to load the class with weak symbols from just `HumanStateProvider` and ignoring the rpc server from `HumanDynamicEstimator`. For further reference : https://github.com/robotology/gazebo-yarp-plugins/issues/105#issuecomment-52401898 https://en.wikipedia.org/wiki/Weak_symbol?oldformat=true#Shared_example 

As a quick fix, renaming the class names to `HumanStateProviderCmdParser` and `HumanDynamicsEstimaorCmdParser` fixes the problem and I can see the correct options from both the rpc ports.

[image]

An ideal way to proceed with handling the rpc functionality for various devices in `HDE` is to either use a correct namespace or using a proper thrift file https://github.com/robotology/wearables/blob/master/msgs/thrift/XsensSuitControlService.thrift and attaching the rpc port to a server https://github.com/robotology/wearables/blob/master/wrappers/IXsensMVNControl/src/IXsensMVNControlWrapper.cpp#L71

@diegoferigo please add any details you see missing and relevant. 

@lrapetti @claudia-lat @traversaro @DanielePucci

diegoferigo, 2019-11-20:
> An ideal way to proceed with handling the rpc functionality for various devices in HDE is to either use a correct namespace or using a proper thrift file [...]

More correctly, without changing the RPC logic, the solution would be to differentiate the two `CmdParser` symbols. There are (at least) two ways:

1. Put the class in a unique namespace. In this way, the scope provided by the two different namespaces would make the two symbols dyfferent.
1. Since both classes already use the pimpl idiom, you could take advantage of that class (which symbol is already scoped since it belongs to the device class) and implement there RPC method. I think we do something similar in Wearables.

As you said using a thrift message would be another solution, though it would require more work and complexity (a new thrift file and the process to autogenerate and build the message class) without adding much value.

lrapetti, 2019-11-22:
> 2\. Since both classes already use the pimpl idiom, you could take advantage of that class (which symbol is already scoped since it belongs to the device class) and implement there RPC method. I think we do something similar in Wearables.

I have implemented this solution with https://github.com/robotology/human-dynamics-estimation/pull/156/commits/7e783c445c4e4c9f447567c48f22cba21b30ea0d

lrapetti, 2019-11-25:
The PR has been merged in the feature branch where we implemented the rpc ports.

Closing.

closed 2019-11-25 by lrapetti