gbionics/human-dynamics-estimation #162: Compilation fails against YARP 3.2.101 (upcoming 3.3)

traversaro, 2019-11-26:
With the following error: 
~~~
2019-11-26T08:34:43.1479473Z /home/runner/work/robotology-superbuild/robotology-superbuild/robotology/human-dynamics-estimation/wrappers/HumanStateWrapper/HumanStateWrapper.cpp:11:10: fatal error: thrift/HumanState.h: No such file or directory
2019-11-26T08:34:43.1479915Z  #include "thrift/HumanState.h"
~~~

I guess the issue is similar to https://github.com/robotology/wearables/issues/56 that was fixed by https://github.com/robotology/wearables/pull/65 . 

cc @DanielePucci

lrapetti, 2019-11-26:
PR towards `master` opened in https://github.com/robotology/human-dynamics-estimation/pull/163

lrapetti, 2019-11-26:
PR merged in master (https://github.com/robotology/human-dynamics-estimation/pull/163).
For the devel branch I am actually not experiencing any failure during the compilation, maybe thanks to @kouroshD changes in https://github.com/robotology/human-dynamics-estimation/pull/155.

@traversaro can you confirm the problem was only with HumanDynamicsEstimation master branch?

traversaro, 2019-11-26:
> @traversaro can you confirm the problem was only with HumanDynamicsEstimation master branch?

I fixed some other issues in unstable branches (i.e. https://github.com/robotology/idyntree/issues/599) and now I rested the build, you can follow it https://github.com/robotology/robotology-superbuild/pull/306/checks .

traversaro, 2019-11-26:
> > @traversaro can you confirm the problem was only with HumanDynamicsEstimation master branch?
> 
> I fixed some other issues in unstable branches (i.e. [robotology/idyntree#599](https://github.com/robotology/idyntree/issues/599)) and now I rested the build, you can follow it https://github.com/robotology/robotology-superbuild/pull/306/checks .

The unstable branches seems to be compiling fine in https://github.com/robotology/robotology-superbuild/pull/306 .

lrapetti, 2019-11-26:
> The unstable branches seems to be compiling fine in [robotology/robotology-superbuild#306](https://github.com/robotology/robotology-superbuild/pull/306) .

Great, thanks @traversaro. I am closing this issue.

closed 2019-11-26 by lrapetti