gbionics/human-dynamics-estimation #167: Investigate why QP solver in IK does not work for some robots

kouroshD, 2019-12-10:
IK of the robots other than iCub model with QP option did not work fine, we should investigate and resolve this problem.

lrapetti, 2019-12-11:
Which type of behaviour did you experience? And did you experience problem also with the human model? Because I remember I was having problem sometimes with the human model, in particular sometimes all the joints were fixed to zero

kouroshD, 2019-12-11:
yes, the joints were fixed to zero, it was mainly the robot models. I should investigate that.

traversaro, 2020-01-31:
As usual, it would be great to get if the problem is in the generated QP problem, on if in the QP solver itself, by saving the instance on a file. If the QP problem is fine and the issue is in the solver, it may be related  to https://github.com/oxfordcontrol/osqp/issues/151 .

lrapetti, 2020-01-31:
I would like to underline the fact that the problem showed up recently, and before we used QP solver without problem for a while.  We should then also check with older implementation, otherwise it might be reltated to some regression in `osqp`.

traversaro, 2020-01-31:
> I would like to underline the fact that the problem showed up recently, and before we used QP solver without problem for a while. We should then also check with older implementation, otherwise it might be related to some regression in `osqp`.

Which version of osqp are you using? The version in the robotology-superbuild has been fixed for a long time, before we bumped it in https://github.com/robotology/robotology-superbuild/pull/330 .

lrapetti, 2020-01-31:
> Which version of osqp are you using?

I just checked and I am on `0.5` that dates back to feb2019, so probably the regression is not related to that.

lrapetti, 2020-10-20:
CC @CarlottaSartore

traversaro, 2020-10-20:
If this is still happening, the first step is probably try to find a way to always dump the QP instance if a given QP instance fails, so if it ever happens, we have some more data to evaluate what is going.

CarlottaSartore, 2020-10-20:
When I am using the QP solver I got all the joints value to zero. I moved to  Cholesky decomposition and everything works fine. 
Please notice that for now, I do not have any limits (e.g. joints limits or velocity one) so the problem could be ill-defined or some parameters not correctly set. Despite this, I do not have any error but simply the QP output to zero.

kouroshD, 2020-10-20:
I can work on this issue, on November. If this is blocking for some experiments, I can give some support to understand where the problem is coming.

lrapetti, 2020-10-20:
Just add a couple of extra details:
- when we say the output is zero, it is not sharp zero but there is some numerical noise (we should check however the exact output of the QP, the noise might be added at a different a step of the device such as the integration).
- the QP apparently is not failing since we never enter [this](https://github.com/robotology/human-dynamics-estimation/blob/master/devices/HumanStateProvider/InverseVelocityKinematics/InverseVelocityKinematics.cpp#L376-L381) error condition.

traversaro, 2020-10-20:
> When I am using the QP solver I got all the joints value to zero. I moved to Cholesky decomposition and everything works fine.
> Please notice that for now, I do not have any limits (e.g. joints [...]

@CarlottaSartore are you able to save an instance (i.e. the cost function and constraints) of the failing problem?

CarlottaSartore, 2020-10-20:
Yes sure,  the qp call is made in [this ](https://github.com/robotology/human-dynamics-estimation/blob/master/devices/HumanStateProvider/InverseVelocityKinematics/InverseVelocityKinematics.cpp) file correct ? @lrapetti

lrapetti, 2020-10-20:
> Yes sure, the qp call is made in [this ](https://github.com/robotology/human-dynamics-estimation/blob/master/devices/HumanStateProvider/InverseVelocityKinematics/InverseVelocityKinematics.cpp) file correct ? @lrapetti

that's right.

lrapetti, 2020-11-17:
Today I have experienced again this problem when using the human model.

In particular I have observed that with the same version of `human-dynamics-estimation` and `Osqp-eigen`, I am getting the output from my machine (`MacOS 10.14.6`), but I am getting all zeros from a different machine (`Ubuntu 20.04`).

CarlottaSartore, 2020-11-17:
@lrapetti  if you are currently working on that can you save  [this](https://github.com/robotology/human-dynamics-estimation/issues/167#issuecomment-712712840) data ?

lrapetti, 2020-11-17:
> @lrapetti if you are currently working on that can you save [this](https://github.com/robotology/human-dynamics-estimation/issues/167#issuecomment-712712840) data ?

I am not sure which kind of information we can get, from [`osqp-eigen` solver interfaces](https://robotology.github.io/osqp-eigen/doxygen/doc/html/classOsqpEigen_1_1Solver.html) I don't see those information exposed, but maybe @GiulioRomualdi can give some pointers to how to retrieve those information.

traversaro, 2020-11-17:
> I am not sure which kind of information we can get, from [`osqp-eigen` solver interfaces](https://robotology.github.io/osqp-eigen/doxygen/doc/html/classOsqpEigen_1_1Solver.html) I don't see those [...]

You can log the data that you feed inside the QP solver (i.e. the problem) with the osqp-eigen interface, but if for any reason you want to access it after you set them, you can use the `workspace()` method that gives you the osqpworkspace that with the osqpdata should contain everything. For saving data https://github.com/dic-iit/matio-cpp/ could be useful.

kouroshD, 2021-11-11:
I dedicated some time today to fix this old problem:
Here is the commit: [cb63f016fd3cb9840651f60686042fea696ba419](https://github.com/robotology/human-dynamics-estimation/commit/cb63f016fd3cb9840651f60686042fea696ba419)

The problem was related to parameters of the constraints: `k_u` and `k_l` which were not set when we did not have custom constraint matrix, to be more clear, those parameters were read from the config file, but it was set in the `InverseVelocityKinematics` class, so they had their initial value of zero. This was done, since those parameters were set with the same method which was setting the custom constraint matrices and vectors. So, I made them separate and check the results.

kouroshD, 2021-11-11:
Here is a validation of the proposed solution and investigation:

https://user-images.githubusercontent.com/17707730/141313528-03b45202-2738-4836-af94-1b4a6a745d89.mp4

kouroshD, 2021-11-11:
So, with this PR I am concluding this issue: https://github.com/robotology/human-dynamics-estimation/pull/272

lrapetti, 2021-11-12:
> So, with this PR I am concluding this issue: #272

The PR was merged. We can wait for the final release of the fix in the `master` branch (PR: https://github.com/robotology/human-dynamics-estimation/pull/265)

lrapetti, 2021-11-15:
> > So, with this PR I am concluding this issue: #272
> 
> The PR was merged. We can wait for the final release of the fix in the `master` branch (PR: #265)

Merged to master! We can close it

closed 2021-11-15 by lrapetti