gbionics/human-dynamics-estimation #170: Add a wiki page on How to Run HDE Demo

Yeshasvitvs, 2020-01-27:
Following the issue on adding a [wiki page](https://github.com/dic-iit/component_andy/issues/155) for the usage of HDE, one missing point is the wiki page on how to run HDE demo. This issue will track the details on adding a new wiki page that will provide details to run HDE demo.

@lrapetti @kouroshD @claudia-lat @DanielePucci

Yeshasvitvs, 2020-03-11:
A [new wiki page](https://github.com/robotology/human-dynamics-estimation/wiki/Running-HDE-Related-Demos) for running HDE related demos is added.

@lrapetti @kouroshD please double check and feel free to update it if you come across some inconsistencies. Thanks

Yeshasvitvs, 2020-03-16:
@claudia-lat I moved this issue to review and thought to close it in the next iteration meeting.
Given the change in organization, i think we can close this issue. Let me know.

DanielePucci, 2020-03-16:
Agreed @Yeshasvitvs

closed 2020-03-16 by Yeshasvitvs