gbionics/human-dynamics-estimation #171: Windows compilation failure on master since today

traversaro, 2020-01-30:
Today there has been a regression on the master branch (apparently the robotology-superbuild uses the master branch as both stable and unstable tags for wearables and human-dynamics-estimation, and this may  be a problem as we catch certain issues only when they land in master). In particular, Windows compilation is now failing with this error: 
~~~
2020-01-30T11:43:43.6865797Z     Generating Code...
2020-01-30T11:43:46.7622726Z D:\a\robotology-superbuild\robotology-superbuild\robotology\human-dynamics-estimation\devices\HumanStateProvider\InverseVelocityKinematics\InverseVelocityKinematics.cpp : fatal error C1128: number of sections exceeded object file format limit: compile with /bigobj [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\human-dynamics-estimation\devices\HumanStateProvider\HumanStateProvider.vcxproj] [D:\a\robotology-superbuild\robotology-superbuild\build\human-dynamics-estimation.vcxproj]
2020-01-30T11:43:46.9773425Z     Building Custom Rule D:/a/robotology-superbuild/robotology-superbuild/robotology/human-dynamics-estimation/publishers/HumanStatePublisher/CMakeLists.txt
2020-01-30T11:43:47.0788678Z     HumanStatePublisher.cpp
~~~

See https://github.com/robotology/robotology-superbuild/runs/416976325 for a related failure. 
You  can read more in detail about the error in https://docs.microsoft.com/en-us/cpp/error-messages/compiler-errors-1/fatal-error-c1128?view=vs-2019 , even if I am not sure what is  actually creating it, as the `HumanStatePublisher.cpp` does not seems to be so big. However, to fix it is should be sufficient to just pass the `/bigobj` option to the compiler when compiling the compilation units of that library, by adding: 
~~~
if(MSVC)
  target_compile_options(HumanStatePublisher PRIVATE /bigobj)
endif()
~~~

traversaro, 2020-01-31:
@lrapetti @Yeshasvitvs This regression is quite critical for the whole robotology-superbuild CI system. If you are ot able to work on it, could you at least comment on when you think you can work on this, so that we can organize how to deal with this, for example disabling this software in the CI? 

Thanks a lot.

lrapetti, 2020-01-31:
> @lrapetti @Yeshasvitvs This regression is quite critical for the whole robotology-superbuild CI system. If you are ot able to work on it, could you at least comment on when you think you can work on [...]

If the fix concern just adding the few line you mentioned above I can do it, but unfortunately I am not sure I will be able to do any actual test on Windows today.

Anyway is the link in your comment above correct:
> See https://github.com/robotology/robotology-superbuild/runs/416976325 for a related failure.

because I see an error related to `robots-configuration` there

traversaro, 2020-01-31:
> If the fix concern just adding the few line you mentioned above I can do it, but unfortunately I am not sure I will be able to do any actual test on Windows today.

Even if that does not fix it, "perhaps" failing the compilation is  for sure better then "for sure" failing the compilation. Furthermore, once the fix is pushed to master, we can see immediately if it works via the jobs of the robotology-superbuild are fixed.

> Anyway is the link in your comment above correct:
> 
> > See https://github.com/robotology/robotology-superbuild/runs/416976325 for a related failure.
> 
> because I see an error related to `robots-configuration` there

Yes. The messages related to `robots-configuration` at the end are just warnings, the error related to human-dynamics-estimation are at line 42087 (hint: search for "fatal error").

lrapetti, 2020-01-31:
@traversaro please check https://github.com/robotology/human-dynamics-estimation/pull/172 if it is what you were suggesting

traversaro, 2020-01-31:
@lrapetti it is!

traversaro, 2020-02-01:
Unfortunatly the failure is also in other devices, see https://github.com/robotology/robotology-superbuild/pull/338/checks?check_run_id=420219032  : 
~~~
2020-01-31T23:42:11.4468211Z D:\a\robotology-superbuild\robotology-superbuild\robotology\human-dynamics-estimation\devices\HumanStateProvider\InverseVelocityKinematics\InverseVelocityKinematics.cpp : fatal error C1128: number of sections exceeded object file format limit: compile with /bigobj [D:\a\robotology-superbuild\robotology-superbuild\build\robotology\human-dynamics-estimation\devices\HumanStateProvider\HumanStateProvider.vcxproj] [D:\a\robotology-superbuild\robotology-superbuild\build\human-dynamics-estimation.vcxproj]
~~~

lrapetti, 2020-02-03:
> Unfortunatly the failure is also in other devices, see https://github.com/robotology/robotology-superbuild/pull/338/checks?check_run_id=420219032 :
>
> ```
> 2020-01-31T23:42:11.4468211Z [...]

PR adding to `HumanStateProvider` and `HumanDynamicsEstimator` (https://github.com/robotology/human-dynamics-estimation/pull/174)

lrapetti, 2020-02-03:
Fixed https://github.com/robotology/robotology-superbuild/issues/339

closed 2020-02-03 by lrapetti