gbionics/human-dynamics-estimation #175: Update HumanStateWrapper to provide rate of change of momentum values and visualize through yarpscope

Yeshasvitvs, 2020-02-03:
We use [yarpscope](https://www.yarp.it/yarpscope.html) as a handy tool to visualize real-time data of external wrench estimation at the hands. It will be use to have also the data of rate of change of momentum visualized alongside the estimates of wrench. This will help us in understanding the effect of the new constraints in the first task of stack of tasks berdy.  This issue tracks the details related to updating `HumanStateWrapper` to stream rate of change of momentum values through yarp ports.

@lrapetti @kouroshD @claudia-lat @DanielePucci

Yeshasvitvs, 2020-02-03:
One obvious problem with data coming from outputPort of `HumanStateWrapper` is it is a complex yarp bottle with strings (for joint names) and joint positions and velocities. Unfortunately, yarpscope does not have the feasibility to handle such ports (@traversaro can you please correct me if I am wrong on this).  

A quick and dirty work around is to add a new yarp port to stream just the values of rate of change of momentum so that we can have yarpscope xml file for online visualization. 

@lrapetti @diegoferigo @traversaro

traversaro, 2020-02-03:
> One obvious problem with data coming from outputPort of `HumanStateWrapper` is it is a complex yarp bottle with strings (for joint names) and joint positions and velocities. Unfortunately, yarpscope [...]

Not out of the box, but see https://github.com/robotology/yarp/issues/1693#issuecomment-574683563 for a related workflow.

> A quick and dirty work around is to add a new yarp port to stream just the values of rate of change of momentum so that we can have yarpscope xml file for online visualization.

An alternative is to have a bridge program that reads from the "complex" port and streams the plain vector.

Yeshasvitvs, 2020-02-03:
> An alternative is to have a bridge program that reads from the "complex" port and streams the plain vector.

@traversaro related to it https://github.com/robotology/human-dynamics-estimation/issues/151#issuecomment-581398313

Yeshasvitvs, 2020-02-04:
In view of https://github.com/robotology/human-dynamics-estimation/issues/176, this issue is not longer relevant. So, closing this issue.

CC @lrapetti @kouroshD @claudia-lat @DanielePucci

closed 2020-02-04 by Yeshasvitvs