gbionics/human-dynamics-estimation #177: Bug: Fix wrong number of channels exposed through IAnalogSensor interface from HumanDynamicsEstimator

Yeshasvitvs, 2020-02-05:
In the implementation of of [IAnalogSensor](https://github.com/robotology/human-dynamics-estimation/blob/6eeadb592cfc63f795efd2d6410d926b2e284dbd/devices/HumanDynamicsEstimator/HumanDynamicsEstimator.cpp#L2555)  interface in `HumanDynamicsEstimator` device, the [number of channels](https://github.com/robotology/human-dynamics-estimation/blob/6eeadb592cfc63f795efd2d6410d926b2e284dbd/devices/HumanDynamicsEstimator/HumanDynamicsEstimator.cpp#L2581) exposed are incorrect i.e. is exposes only estimated wrench channels which is `24`. The `IAnalogSensor` implemented in `HumanDynamicsEstimator` device is designed to exposed both the offset removed wrench measurements (i.e., 4  wrench sources * 6 = 24 Channels) + estimated wrenches (4 links * 6 = 24 Channels). 

CC @lrapetti

lrapetti, 2020-02-05:
> device is designed to exposed both the offset removed wrench measurements (i.e., 4 wrench sources * 6 = 24 Channels)

In general we might also avoid exposing the **offset removed wrench measurements**, it was useful mainly for debugging purpose in the past (maybe also future)

Yeshasvitvs, 2020-02-06:
@lrapetti till we have a stable behavior with the external wrench estimation at the hands we will be using offset removed wrench measurements for debugging. So, for being consistent with the values exposed and the number of channels exposed to through `IAnalogSensor` interface I names this issue as a bug, but you are right.

Yeshasvitvs, 2020-02-07:
This is fixed in PR https://github.com/robotology/human-dynamics-estimation/pull/178

Closing

closed 2020-02-07 by Yeshasvitvs