gbionics/human-dynamics-estimation #179: Update HumanDataCollector device with logging using matio library

Yeshasvitvs, 2020-02-07:
Following the development of `HumanDataCollector` https://github.com/robotology/human-dynamics-estimation/issues/176, a nice feature is to have the data logging possibility in preferably matlab `.mat` files as discussed in https://github.com/robotology/human-dynamics-estimation/issues/151. This issue tracks the details of updating the `HumanDataCollector` device with the feature to log human data to a `.mat` file using [matio library](https://github.com/tbeu/matio).

CC @lrapetti @diegoferigo @traversaro @DanielePucci

diegoferigo, 2020-02-07:
I think that the first step could be defining the desired data structure

lrapetti, 2020-02-07:
> I think that the first step could be defining the desired data structure

Maybe some structure used by @claudia-lat can be a starting point

Yeshasvitvs, 2020-02-07:
> Maybe some structure used by @claudia-lat can be a starting point

At this point I would like to not go in depth on the data structure used in [MAPEst](https://github.com/claudia-lat/MAPest) and I prefer to focus on getting something concrete quickly :sweat_smile: 

> I think that the first step could be defining the desired data structure

At this moment, I can find the following two sources as reference for using `matio` library

- https://github.com/NJannasch/matio-examples
- https://sourceforge.net/projects/matio/files/matio/1.5.2/matio_user_guide.pdf/download

From what I understand, an starting point is to have the following data structure:

-  `data` cell array with the following cell elements
    - `time` numeric array of dimension `s x 1`  (where `s` indicates the number of samples)
    - `basePose numeric array of dimension `s x 7` (angular part expressed in Quaternions) 
    - `baseVelocity` numeric array of dimension `s x 6` (angular part expressed in RPY) 
    - `comPosition numeric array of dimension `s x 3`
    - `comVelocity` numeric array of dimension `s x 3`
    - `comProperAccelerationInBaseFrame` numeric array of dimension `s x 6`
    - `comProperAccelerationInWorldFrame` numeric array of dimension `s x 6`
    - `stateJointNames` string array of dimension `1 x n` **[TO BE CHECKED]**
    - `jointPositions` numeric array of dimension `s x n` (where `n` indicates the degrees of freedom)
    - `jointVelocities` numeric array of dimension `s x n`
    - `wrenchMeasurements` cell array with `w` cell elements corresponding to the number of wrench sources considered:
        - Each cell element is a numeric array of dimension `s x 6`
    - `wrenchMeasurements` cell array with `2 * w` cell elements corresponding to the number of wrench sources considered:
        - Each cell element is a numeric array of dimension `s x 6`
    - `jointTorques` numeric array of dimension `s x n` 
    - `dynamicsJointNames` string array of dimension `1 x n` **[TO BE CHECKED]**

PS: the variables  `stateJointNames` and `dynamicsJointNames` should ideally contain the same number and order for the joint names. However, it is better to also log this data, as these two variables come from different interfaces implemented in different devices.

diegoferigo, 2020-02-07:
I want also to drop https://github.com/GiulioRomualdi/YarpMatLogger from which you can get some matio logic

cc @GiulioRomualdi

lrapetti, 2020-02-07:
> I want also to drop https://github.com/GiulioRomualdi/YarpMatLogger from which you can get some matio logic
> 

In that case I think @GiulioRomualdi was using [`matlogger2`](https://github.com/ADVRHumanoids/MatLogger2) (that is based on `matio` so can be also a starting point)

traversaro, 2020-02-08:
Another interesting lib: https://github.com/tesch1/eigen-matio . Note that `matlogger2` is nice, but it has some small parts (that can be easily be fixed) that currently prevent its use on Windows.

Yeshasvitvs, 2020-02-17:
A PR https://github.com/robotology/human-dynamics-estimation/pull/181 that adds the matio logging feature to `HumanDataCollector` device is open.  Different from the data structure I initially planned in https://github.com/robotology/human-dynamics-estimation/issues/179#issuecomment-583397153, I updated it to the following:

##### String Variables

The following four variables are vector of string variables:

- `stateJointNames`
- `wrenchMeasurementSourceNames`
- `wrenchEstimateSourceNames`
- `dynamicsJointNames`

These four variables are metadata and they are stored as mat cell variables of the same names with size `rows x 1`, with the cell elements corresponding to the strings.

[image]

##### Double Variables

The rest of the variables are various measurements values and they are stored into a mat struct variable with the name `data` and filed names corresponding to the names of the quantities. Each struct field is a numeric array of size `cols x samples`

[image]

Thanks to @traversaro and @diegoferigo for the support on addressing some of the issues related to this feature.

Yeshasvitvs, 2020-03-03:
At the moment, the matio-logging is done on termination of HDE. Saving to `.mat` file happens in the `detach()` method of `HumanDataCollector` device. This is not ideal when the data dump has of a big size. So, ideally we need to have the matio-logging feature to update the data constantly to the mat file, while HDE devices are running. However, as discussed in the iteration meeting yesterday with @lrapetti @kouroshD @claudia-lat, the matio-logging is stable for experiments over a duration of 10 minutes and `.mat` files of size > 150 MB. So, for the time being, we can proceed with merging the associated PR https://github.com/robotology/human-dynamics-estimation/pull/181, and implement the feature of parallel logging later.

claudia-lat, 2020-03-09:
@yeshi please close this issue.

Yeshasvitvs, 2020-03-09:
The PR https://github.com/robotology/human-dynamics-estimation/pull/181 has been merged. 
As discussed in https://github.com/robotology/human-dynamics-estimation/issues/179#issuecomment-593951539, run time logging through parallel buffers is a task to be done for future requirements of experiments needing more than 15 mins of data dumping, or higher file size (which will be the result if the HumanDataCollector device period is low)

For the time being this issue can be closed.

closed 2020-03-09 by Yeshasvitvs