gbionics/human-dynamics-estimation #180: Investigate and fix memory leak in human dynamics estimation code base

Yeshasvitvs, 2020-02-11:
Give the code base of [human-dynamics-estimation](https://github.com/robotology/human-dynamics-estimation) repo went through a major refactoring for V2.0.0 and the code review was not consistent, there are a lot memory leaks which sometimes lead to segmentation fault on terminating `yarprobotinterface`. I ran `valgrind` with `yarp`, `iDynTree` and `human-dynamics-estimation` in `debug` mode and the output log is stored at https://gist.github.com/Yeshasvitvs/aab49b48ba94dda1efada6e6e760ef49. `valgrind` is ran with the following options:

```
--tool=memcheck --gen-suppressions=all --track-origins=yes --show-reachable=yes --leak-check=full
```

 In this issue we will track the details on fixing these memory leaks to improve the code.

@lrapetti @kouroshD @traversaro

DanielePucci, 2020-02-15:
Hi @Yeshasvitvs, the issues in the Icebox should not belong to any milestone.

If you believe to process this issue in the next to weeks, leave the milestone and move the issue to the Backlog 

CC
@claudia-lat

lrapetti, 2023-10-19:
Segmentation fault is no longer happening, I would proceed closing this issue.

closed 2023-10-19 by lrapetti