gbionics/human-dynamics-estimation #186: Clarify software to download in the ESD Can docs

traversaro, 2020-02-20:
Following the docs in https://github.com/robotology/human-dynamics-estimation/wiki/Set-up-Machine-for-running-HDE#usb-can-2 the instructions say to download the `CAN_SDK_V***` and `can_usb2_win64_***` files, however those files cannot be found in the linked page https://esd.eu/en/products/can-usb2 . 

It would be great to have some more direct links to the required files. 

cc @kouroshD @lrapetti

traversaro, 2020-02-20:
cc @fiorisi @marcoaccame

kouroshD, 2020-02-21:
If I am not wrong the user should login in order to download the files. maybe @lrapetti  can elaborate more on that.

traversaro, 2020-02-21:
Perhaps also @isorrentino can know something about it.

isorrentino, 2020-02-21:
@traversaro The links that I put in the [WIKI of element_ftSk_Shoes](https://github.com/dic-iit/element_ftSk_Shoes/wiki/How-to-install-ftShoes#install-usb-can-drivers) are:
* USB-CAN mini https://drive.google.com/open?id=1dWAg3u8jlVcYCuPBcv4ZXflj4NkHC_EO
* USB-CAN2 https://drive.google.com/file/d/1_pDrmburFcgtFZhKYgbDvwucj9HntyEm/view?usp=sharing

lrapetti, 2020-02-21:
> Probably in https://github.com/robotology/human-dynamics-estimation/wiki/Set-up-Machine-for-running-HDE#usb-can-2 you can use the same links.

since `human-dynamics-estimation` is a public repo we can not put the link to the drive. That's why we just put a general description of the required driver.

lrapetti, 2020-02-21:
> @traversaro The links that I put in the [WIKI of element_ftSk_Shoes](https://github.com/dic-iit/element_ftSk_Shoes/wiki/How-to-install-ftShoes#install-usb-can-drivers) are:
> 
> * USB-CAN mini https://drive.google.com/open?id=1dWAg3u8jlVcYCuPBcv4ZXflj4NkHC_EO
> * USB-CAN2 https://drive.google.com/file/d/1_pDrmburFcgtFZhKYgbDvwucj9HntyEm/view?usp=sharing
> 
> Probably in https://github.com/robotology/human-dynamics-estimation/wiki/Set-up-Machine-for-running-HDE#usb-can-2 you can use the same links.

probably we should also delete the links from this comment

traversaro, 2020-02-21:
Beside the archive problem, do we have the URL for downloading those files from the ESD  website?

traversaro, 2020-02-21:
Ok, I think that CAN SDK (that is the only one actually required for compilation) can be downloaded  after registration at  https://esd.eu/en/software-downloads/27425 (linked from https://esd.eu/en/products/can-sdk), but I don't know where the USB-CAN2 and USB-CAN mini driver used by @isorrentino can be downloaded.

Yeshasvitvs, 2020-02-21:
> Beside the archive problem, do we have the URL for downloading those files from the ESD website?

The last time I did this was in Oct 2018 at Xsens office and even after the registration, the download option was not available. In the end I got the drivers from @claudia-lat

traversaro, 2020-02-21:
> > Beside the archive problem, do we have the URL for downloading those files from the ESD website?
> 
> The last time I did this was in Oct 2018 at Xsens office and even after the registration, the download option was not available. In the end I got the drivers from @claudia-lat

If the driver is not available from the website and can only be provided directly from ESD, we should mention it in the docs, as otherwise for any user following the docs it will be impossible to "Download the proper drivers from the ESD electronics" as it is currently written in the docs.

traversaro, 2020-02-25:
So, just to double check: expect from the CAN SDK, the drivers for the ESD CAN boxes are not available from the website, and instead are available from the CD shipped with the boxes? 
@Yeshasvitvs @lrapetti (Tagging you as you are the mantainers)

lrapetti, 2020-02-25:
> and instead are available from the CD shipped with the boxes?
> @Yeshasvitvs @lrapetti (Tagging you as you are the mantainers)

Honestly I don't know when/how we got them. I have always used the links that @isorrentino was pointing.

claudia-lat, 2020-02-25:
I can reply to this question.  Few notes:
- The drivers come with the physical ESD-CAN in this form
[image]

- In the IIT storage `//storage05.icub.iit.local/repository/`some old versions of the drivers are stored.
[image]

traversaro, 2020-02-25:
Great, thanks @claudia-lat !

lrapetti, 2020-02-25:
I was planning to change the wiki with the following

> - Before proceeding, check if you have the availability of the proper drivers from the [ESD electronic](https://esd.eu/en/products/can-usb2) that are shipped with the can-usb2 device. You should have the following two files:

(I can not do it at the moment since github is crashing when trying to change the wiki)

traversaro, 2020-02-25:
Perfect, thanks @lrapetti !

claudia-lat, 2020-11-23:
CC @lrapetti

lrapetti, 2020-11-23:
> I was planning to change the wiki with the following
>
>
> (I can not do it at the moment since github is crashing when trying to change the wiki)

Done. Hope now it is more clear, closing.

closed 2020-11-23 by lrapetti