gbionics/human-dynamics-estimation #187: Test HumanStateProvider after rebasing feature/visualize-berdy-estimated-wrench to devel

Yeshasvitvs, 2020-02-20:
Following the PR https://github.com/robotology/human-dynamics-estimation/pull/164 by @lrapetti , major changes are made to `HumanStateProvider` device to add the feature of secondary calibration. Also, `HumanStateProvider` device was updated in parallel with base offset removal features https://github.com/robotology/human-dynamics-estimation/pull/153. 

For the sake of aligning the branches, I rebased  [feature/visualize-berdy-estimated-wrench](https://github.com/robotology/human-dynamics-estimation/tree/feature/visualize-berdy-estimated-wrench) onto the `devel` branch after PR  https://github.com/robotology/human-dynamics-estimation/pull/164 was merged. Clearly, there were a lot of conflicts in `HumanStateProvider.cpp` file and I managed to fix them. I ran HDE and the rviz visualization seems to be fine. However, I did not test all the rpc features related to base offset removal and secondary calibration. So, I pushed changes after rebasing to a new branch https://github.com/Yeshasvitvs/human-dynamics-estimation/tree/test/devel-rebase that needs to be tested. 

@lrapetti please test this branch and most importantly try to double check and test the commit https://github.com/Yeshasvitvs/human-dynamics-estimation/commit/4a60f2ae0f80a383fb1dfe528c0d2296a0a709c1.

lrapetti, 2020-03-16:
Hi @Yeshasvitvs, I am failing in compiling this branch with the following:
```
/Users/lorenzorapetti/Software/robotology-superbuild/robotology/human-dynamics-estimation/devices/HumanDynamicsEstimator/HumanDynamicsEstimator.cpp:1879:34: error: 
      no matching member function for call to
      'updateEstimateInformationFloatingBase'
        pImpl->berdyData.solver->updateEstimateInformationFloatingBase(p...
        ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/lorenzorapetti/Software/robotology-superbuild/build/install/include/iDynTree/Estimation/BerdySparseMAPSolver.h:75:14: note: 
      candidate function not viable: requires 5 arguments, but 6 were provided
        void updateEstimateInformationFloatingBase(const JointPosDoubleA...
             ^
/Users/lorenzorapetti/Software/robotology-superbuild/build/install/include/iDynTree/Estimation/BerdySparseMAPSolver.h:81:14: note: 
      candidate function not viable: requires 7 arguments, but 6 were provided
        void updateEstimateInformationFloatingBase(const Transform& base...
             ^
2 errors generated.
```
I guess I'm not using the correct branch of `wearables`. I am currently using [`idyntree-hde-fork/feature/stack-of-tasks-berdy`](https://github.com/dic-iit/idyntree-hde-fork/tree/feature/stack-of-tasks-berdy) is it correct or should I use another branch

Yeshasvitvs, 2020-03-16:
@lrapetti idyntree branch is correct but may be the commit is not. The recent commits on that branch updated the updateEstimateInformationFloatingBase() with a new argument. Please roll back to the commit previous to this change and try to compile

lrapetti, 2020-03-16:
> @lrapetti idyntree branch is correct but may be the commit is not. The recent commits on that branch updated the updateEstimateInformationFloatingBase() with a new argument. Please roll back to the commit previous to this change and try to compile

I confirm I was able to compile after rolling back to commit https://github.com/dic-iit/idyntree-hde-fork/commit/02d2f3c57ac68d98adeb46f01979d3f6eaf75507

lrapetti, 2020-03-16:
I checked the code and test the command and seem to be answering correctly.
Unfortunately, currently I do not have the possibility to test the visualisation on MacOS so I can not be sure about the behaviour.

Anyway I think you can procede from this rebase and eventually do a final test when we will be merging this branch in devel and if strange behaviours occurs we can fix them later.

claudia-lat, 2020-03-16:
@lrapetti drop here a comment on why we are moving this issue into Backlog.

lrapetti, 2020-03-16:
> @lrapetti drop here a comment on why we are moving this issue into Backlog.

Even if it seems to be working fine, I cannot visualize the output at the moment so it is better to duble check it in the future. At the moment, checking if this rebase work is not critical since the branch is far from being merged, so we can move it to backlog

claudia-lat, 2020-05-25:
@lrapetti you can test in a Ubuntu lab machine the HumanStateProvider

claudia-lat, 2020-11-02:
@lrapetti please try to understand when doing this test.

lrapetti, 2021-10-14:
Some tests were done in the past, anyway we will need to rebase it again after latest changes. I will close the issue for the time being.

closed 2021-10-14 by lrapetti