gbionics/human-dynamics-estimation #190: Update wrench handling pipeline

Yeshasvitvs, 2020-03-04:
In our present architecture, there are two ways wrench information is propagated:

- [IHumanWrench](https://github.com/robotology/human-dynamics-estimation/blob/master/interfaces/IHumanWrench/IHumanWrench.h) interface 

    - Used in [HumanWrenchProvider](https://github.com/robotology/human-dynamics-estimation/blob/master/devices/HumanWrenchProvider/HumanWrenchProvider.h#L68) device to expose the measured wrench expressed in the link frames
- `IAnalogSensor` interface with ros enabled
    - Used in  `HumanWrenchProvider` device to [expose the measured wrench in link frame](https://github.com/robotology/human-dynamics-estimation/blob/master/devices/HumanWrenchProvider/HumanWrenchProvider.cpp#L650) through ros topics

Ideally, the `IHumanWrench` interface is to be used to stream wrench data between difference devices of the human-dynamics-estimation library. Whereas, `IAnalogSensor` interface should only be used to expose wrench data through ros topics.

However, in some of the quick recent developments, this pipeline has been mixed and we are currently exposing the measured and estimated wrench links expressed in link frame directly through the `IAnalogSensor` interface through ros topics for rviz visualization.

Concerning the ros topics for rviz visualization, the correct architecture is:

- Expose the measured wrench in link frame from `HumanWrenchProvider` device
- Expose the estimated wrench in link frame from `HumanDynamicsEstimator` device

In some of our recent tests related to the first task of stack of tasks implementation, we encountered the need for exposing measured and estimated wrench in the following four different frames:
 
- Link frame
- Base frame
- World frame
- Centroidal frame

The `IHumanWrench` interface needs to be updated accordingly to expose the wrench in difference frames. Ideally,

- The measured wrench in different frames needs to be exposed from `HumanWrenchProvider` device. However, this required the `HumanWrenchProvider` to be attached to `HumanStateProvider` device to get the base transform solution which is required for expressing the measured wrench in difference frames.
- Instead, the `HumanDynamicsEstimation` device is attached to both the `HumanWrenchProvider` device and the `HumanStateProvider` device. So, another ideal architecture is to read the measured wrench expressed in link frames from `HumanWrenchProvider` device and expose them in different frames, along with the estimated wrench in different frames, from the `HumanDynamicsEstimator` device. Furthermore, the `HumanDataCollector` logging device will read all the wrenches from the `HumanDynamicsEstimator` and store the data to a `.mat` file as separate variables, which will be cleaner and easier for matlab plotting.

@lrapetti @prashanthr05  @kouroshD @claudia-lat @DanielePucci

claudia-lat, 2020-03-30:
@Yeshasvitvs drop a comment here and eventually close it.

Yeshasvitvs, 2020-03-30:
The following commits address the wrench handling pipeline in the [test branch](https://github.com/Yeshasvitvs/human-dynamics-estimation/commits/tests/HDE-task1-debug):

- [Update IHumanWrench interface with wrench type and frames enum](https://github.com/Yeshasvitvs/human-dynamics-estimation/commit/0a5554a35f536eacea3bfdc125b2d2e877380785)
- [Add method to express wrench in different frames](https://github.com/Yeshasvitvs/human-dynamics-estimation/commit/8a5921b9be6e415d05dbf4228f618025517378f5)
- [Update HumanDataCollector device wrench logging](https://github.com/Yeshasvitvs/human-dynamics-estimation/commit/541ffc988787814fb6d27c2fac4789c885fa8f0c)
- [Add sum of wrenches map to express wrench sum in different frames](https://github.com/Yeshasvitvs/human-dynamics-estimation/commit/76601018135a2abd79ebc6ba1048f374c3976d07)

@diegoferigo @lrapetti @prashanthr05 @kouroshD @claudia-lat @DanielePucci

Yeshasvitvs, 2020-04-06:
Concerning the SOT Berdy, it is useful to have the measured and estimated wrenches expressed in different frames both from the high priority task (task 1) and low priority task (task 2) dumped to a log file. The task type is more relevant to `HumanDynamicsEstimator` device and having a enumerated type for it in `IHumanWrench` interface is not  really ideal. However, it is a quick implementation and it is done in [Update wrench handling for task 1 and task 2](https://github.com/Yeshasvitvs/human-dynamics-estimation/commit/94ca8299a1e6970efee7046a97ce64757d039f82)

Yeshasvitvs, 2020-08-03:
This feature is implemented in https://github.com/Yeshasvitvs/human-dynamics-estimation/tree/feature/SOT-Berdy-HDE which will be merged to Upstream. So closing this issue.

closed 2020-08-03 by Yeshasvitvs