gbionics/human-dynamics-estimation #197: Cleanup CMake code to install YARP devices

traversaro, 2020-05-24:
I noticed that in this repo manually wrote .ini files for all the devices are still installed, while YARP has gained support to automatically generated them (reducing the possibility of errors) in version 3.2.0 : https://github.com/robotology/yarp/blob/2522239984ef3e6c26939fe740001ec08189a8cb/doc/release/v3_2_0.md#build-system .

For this reason, I suggest to remove all the manually written .ini files that describe a device, and use instead the much more convenient and less error prone automatic generation provided by YARP, see robotology/icub-main#586 for an example pull request.

Yeshasvitvs, 2020-05-25:
CC @lrapetti @kouroshD @claudia-lat @Yeshasvitvs

lrapetti, 2020-05-28:
Done with https://github.com/robotology/human-dynamics-estimation/pull/198

closed 2020-05-28 by lrapetti