gbionics/human-dynamics-estimation #199: Docker support

VladimirIvan, 2020-06-04:
Hi,
I'd like to try the estimation estimation (rather than develop it at this time) but I haven't used YARP and your build/dependency systems before.

Would it be possible to get this software with all its dependencies in a docker image?

Yeshasvitvs, 2020-06-04:
@VladimirIvan Glad to know your interest in using human-dynamics-estimation.

At the moment we do not have any docker image. We will try to create one and update you as soon as possible.

@lrapetti

traversaro, 2021-05-19:
Even if not directly in a Docker image, the `human-dynamics-estimation` is available via the https://github.com/robotology/robotology-superbuild, that itself can be easily installable in a docker image via a few commands. Another option is to use conda packages (see https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md), but this option may be problematic if you want to mix it with apt-provided software.

VladimirIvan, 2021-05-19:
Thank you. This really helps with the setup.

closed 2021-05-19 by VladimirIvan

traversaro, 2021-05-19:
I forgot that if you build from source, to use `human-dynamics-estimation` you need to enable the Human Dynamics profile via the `ROBOTOLOGY_ENABLE_HUMAN_DYNAMICS`  CMake option (https://github.com/robotology/robotology-superbuild/blob/master/doc/cmake-options.md#human-dynamics). If you instead you are familiar with conda, you can install it simply as:
~~~
conda -c conda-forge -c robotology human-dynamics-estimation
~~~
see https://anaconda.org/robotology/human-dynamics-estimation for the available versions.