gbionics/human-dynamics-estimation #200: Update ros launch files xacro usage for ros noetic

Yeshasvitvs, 2020-06-23:
Ubuntu 20.04 does not support ROS Melodic and [it is recommended](http://wiki.ros.org/ROS/Installation) to use ROS Noetic. When trying to launch rviz visualization using the [launch files](https://github.com/robotology/human-dynamics-estimation/tree/master/conf/ros/launch), the following error was thrown:

[image]

On double checking with the [urdf_tutorial display.launch file](https://github.com/ros/urdf_tutorial/blob/master/launch/display.launch), I notice the [usage of xacro](https://github.com/ros/urdf_tutorial/commit/dadba7524ba0d20ecd6d23095f49843c3f8e8d1b#diff-7f8c819cc88e03640153972121b4f8cd) has been updated. So, I changed the usage of xacro in the HDE launch files too and rviz runs correctly. 

We need to check if updating the `xacro` usage on Ubuntu 18.04 will not cause any trouble and update the launch files accordingly. 

CC @lrapetti @kouroshD @claudia-lat

lrapetti, 2020-11-17:
Given that this change should be there since ROS Jade http://wiki.ros.org/xacro#Processing_Order (2015 distribution), I think we should be able to safely update it.

I will open a PR for this change.

claudia-lat, 2020-12-01:
We can close the issue.

closed 2020-12-01 by claudia-lat