gbionics/human-dynamics-estimation #206: Implement Remapper for HumanState interface

lrapetti, 2020-07-24:
This issue has the purpose of giving a bit of background for PR https://github.com/robotology/human-dynamics-estimation/pull/204. 

Let's consider the general scheme for the software infrastructure at https://github.com/robotology/human-dynamics-estimation/blob/master/misc/hde_scheme.png. We can observe that at the `wearable` level we can run different devices in different processes by using yarp network and connecting the interfaces using `wrapper`/`remapper` devices that stream and read the interface data trough yarp ports.

In the case of the `human-dynamics-estimation` we have to run the whole block in a single `yarprobotinterface`. This is fine in most of the cases, but there might be some cases in which we want to distribute the processes execution in different processes, or we need to run the devices in different moment (e.g. running `HumanStateProvider` and `RobotPositionController`). In order to do so we can use the same strategy used in `wearables` and connect the devices using `wrapper`\`remapper` devices. 

Since the `wrapper` device for the `IHumanState` interface is already there, we just need to develop the remapper device.

claudia-lat, 2020-09-08:
CC @lrapetti

lrapetti, 2020-09-18:
Merged (https://github.com/robotology/human-dynamics-estimation/pull/204) closing

closed 2020-09-18 by lrapetti