gbionics/human-dynamics-estimation #209: Understand source of asymmetry when using HumanStateProvider for retargeting on robot

lrapetti, 2020-07-31:
When performing human-to-robot retargeting using real iCub robot (`iCubGenova04`), breakage of the left shoulder cable have been observed often (e.g. see https://github.com/robotology/icub-tech-support/issues/601 or other breakage report on the same repo).

After a first investigation done by @S-Dafarra, the left shoulder cable breakage seems to be related to this specific application and not a general weakness of the robot. Indeed, we should investigate if it is due to any source of asymmetry at the software level.

possible sources of asymmetry to be investigated:
- raw measurements from Xsens sensors
- sensors positioning on the [teleoperation iCub model](https://github.com/robotology/human-dynamics-estimation/blob/master/conf/urdfs/teleoperation_iCub_model_V_2_5.urdf)
- joint constraints asymmetry in the [`RobotStateProvider` configuration file](https://github.com/robotology/human-dynamics-estimation/blob/master/conf/xml/RobotStateProvider_iCub.xml)

@kouroshD @Yeshasvitvs @S-Dafarra

claudia-lat, 2020-09-28:
@kouroshD maybe this issue can be useful for your activity? Would you like to investigate on it? or may we close it?

kouroshD, 2020-09-29:
> After a first investigation done by @S-Dafarra, the left shoulder cable breakage seems to be related to this specific application and not a general weakness of the robot. Indeed, we should investigate if it is due to any source of asymmetry at the software level.

The problem is that, mainly in retargeting scenarios we use the robot arms, so we can be sure about the hardware with this investigation.

> sensors positioning on the teleoperation iCub model

Well, if the sensor positioning was wrong, we could easily see when retargeting the robot. I am not sure if this is the case.

> joint constraints asymmetry in the RobotStateProvider configuration file

If there is an asymmetry in the constraints, then it should be related to hardware. Since, at software level, the left and right arms are acting similarly.

 > raw measurements from Xsens sensors

Well, the rViz visualization was fine as far as I remember during the experiments.

Moreover, we also reduce the maximum allowed shoulder joints velocities, and as far as I remember, we have not had this problem later.
https://github.com/robotology/human-dynamics-estimation/blob/65c3774a92a1a37925fe6cce135639941f4aeb51/conf/xml/RobotStateProvider_iCub_Pole.xml#L70-L73

So @claudia-lat , I am not sure what should we investigate for.

claudia-lat, 2020-10-05:
Please @lrapetti drop a comment here and eventually close the issue.

lrapetti, 2020-10-05:
Given the comment form @kouroshD, we still don't have a clear explanation regarding the occurrence of the breakage on left arm. Anyway, properly setting the joint velocity constraints we did not experience the problem any longer.

For the time being we can close the issue, we can reopen it in case we will need to investigate further.

closed 2020-10-05 by lrapetti