gbionics/human-dynamics-estimation #212: Update yarpmanager retargeting application with all possible module dependencies to ensure correct Rviz model visualization

Yeshasvitvs, 2020-08-24:
Recently we observed that Rviz model visulization was failing when running the [retargeting demo](https://github.com/robotology/human-dynamics-estimation/blob/master/conf/xml/applications/XsensRetargetingVisualization.xml) yarpmanager application. One of the action points to address this issue is to ensure that all the dependencies between different modules are correctly placed, so that before running Rviz model visualization all the data needed for visualization is accessible to Rviz. In this issue will track the details related to adding the module dependencies. 

CC @S-Dafarra @lrapetti @kouroshD

lrapetti, 2021-10-14:
> Any update @lrapetti? Otherwise we can close it.

Dependencies are still missing (see https://github.com/robotology/human-dynamics-estimation/blob/master/conf/xml/applications/HumanDynamicsEstimation-HumanDynamics.xml#L46-L52), I will try to address this task.

lrapetti, 2022-10-19:
`yarpmanager` application for the retargeting is now working and the Rviz visualizer has been removed (see https://github.com/robotology/human-dynamics-estimation/blob/master/conf/xml/applications/HDERetargeting_iCub2_5.xml). I think we can keep using it as it is and close it.

closed 2022-10-19 by lrapetti