gbionics/human-dynamics-estimation #214: Fix CI on MacOS

lrapetti, 2020-10-09:
Currently is failing (see https://travis-ci.org/github/robotology/human-dynamics-estimation/builds/733972242).

It seems to be failing when `brew install eigen`, at the moment I don't have many clues regarding the reason, the error I see is:
```
Error: eigen 3.3.7 is already installed
```

lrapetti, 2020-10-09:
Maybe @traversaro experienced something similar?

traversaro, 2020-10-09:
> 
> 
> Maybe @traversaro experienced something similar?

Apparently brew prints an error if a port request to be installed is already installed but it is not up to date, see https://stackoverflow.com/questions/43619480/upgrade-or-install-a-homebrew-formula for more info. A quick solution may be just to remove eigen from the installed formulas.

traversaro, 2020-10-09:
Back in time on Travis I remember that we had a brewInstallOrUpgrade script exactly for this, but on GitHub Action I never had hte need for something like that (probably because no formulas are pre-installed).

lrapetti, 2020-10-09:
Thanks @traversaro, I will try to adopt `brewInstallOrUpgrade` being currently using Travis for this repo. Still, I don't know why `eigen` appears to be already installed.

traversaro, 2020-10-19:
> Still, I don't know why eigen appears to be already installed.

The Travis macOS image can change from time to time, and it is possible that they decided to pre-install Eigen now.

claudia-lat, 2020-10-26:
The related PR has been merged. 
Closing issue

closed 2020-10-26 by claudia-lat