gbionics/human-dynamics-estimation #222: IK joints output ordered different on different machines

CarlottaSartore, 2020-11-26:
During some experiments, we have noticed that the order in which the joints are outputted by the IK is different in different computers (with the same robotology superuild configuration). 

We temporarily solved this issue by adding a new parameter in the `human-state-wrapper` i.e. the `JointsDesiredOrder` which basically reorders the joints before outputting in the port. (you can find this modification in the branch [wrappers/addVectorWisePublish](https://github.com/robotology/human-dynamics-estimation/tree/wrappers/addVectorWisePublish)). 

C.C. @claudia-lat @lrapetti @Yeshasvitvs @S-Dafarra @gabrielenava @nunoguedelha

Yeshasvitvs, 2020-11-26:
> During some experiments, we have noticed that the order in which the joints are outputted by the IK is different in different computers (with the same robotology superuild configuration).

Thanks for the issue @CarlottaSartore 
Can you please elaborate a bit on where you observed this behavior, is it in the visualization ?

lrapetti, 2020-11-26:
Thanks @CarlottaSartore, I'll try to clarify how things work. The joint order given by the `HumanStateProvider` (and `human-state-wrapper`) depends on the ordering given by `iDynTree::ModelLoader`. I would expect this order be fixed given the same `iDynTree` version (maybe @traversaro has a better insight on this), but for sure in general it is not expected to be a fix order. For this reason, the joints data are exposed together with the joint name list through the [HumanState interface](https://github.com/robotology/human-dynamics-estimation/blob/master/msgs/yarp/thrift/HumanDynamicsEstimation/HumanState.thrift#L23-L34).
The advantage is avoiding hard-coded joint order lists in the `HumanStateProvider` and in the "consumer device" (whoever is reading the joint values), that would require to be maintained together. At the same time, it gives the possibility to any device in the network to be able to interpret the data (besides its OS or `iDynTree` version).

In the past @kouroshD  observed the same "problem" for the `XsensRetargeting` module where we had a mismatch between the model joint order in the laptop where running the retargeting module. In that case, the problem was fixed simply by using the joint name list (https://github.com/robotology/walking-teleoperation/pull/25).

Said that, I agree that in same cases (e.g. from Matlab), it might be helpful to fix this joint order (even if I would say using the joint name list should be always preferred). So feel free to open a PR to add the feature you have implemented.
(Just a small comment on the implementation, I was thinking that, instead of `human-state-wrapper`, this feature might be implemented directly in the `HumanStateProvider` using `loadReducedModelFromFile` option that give the possibility to specify the joint order. In this way we would get for free the possibility to select a subset of joints).

traversaro, 2020-11-26:
> I would expect this order be fixed given the same iDynTree version (maybe @traversaro has a better insight on this), but for sure in general it is not expected to be a fix order.

Yeshasvitvs, 2020-11-26:
> but for sure in general it is not expected to be a fix order. For this reason, the joints data are exposed together with the joint name list through the HumanState interface.

To add to what @lrapetti highlighted, in general client modules that use the information coming from the wrapper needs to ensure the joint names and the corresponding joint values are used.

S-Dafarra, 2020-11-26:
So to summarize, ``HumanStateProvider`` does not enforce any joint order. This leaves the choice to ``iDynTree::ModelLoader`` that might as well return a random list. Hence, I would say a port outputting the joint values might as well throw the lottery numbers.

At this point, I would rather expect the port that outputs only the joint values not to send anything, unless a specific order is first sent to it 🤔 @CarlottaSartore

lrapetti, 2020-11-26:
@S-Dafarra @traversaro probably my comment above was not clear.

We are aware that this joint order is not fixed, that's why we are not not trusting it at all. In fact, the output of `HumanStateProvider` is not a list of joint values, but a structured [interface exposing joint values and joint name list](https://github.com/robotology/human-dynamics-estimation/blob/master/msgs/yarp/thrift/HumanDynamicsEstimation/HumanState.thrift#L23-L34). This interface is then streamed trough a yarp port using the `human-state-wrapper` device, as a structure data:
```
-1 0.000000 2589.299452 (jLeftBallFoot_roty jLeftBallFoot_rotx jLeftAnkle_rotz jLeftAnkle_roty jLeftAnkle_rotx jLeftKnee_rotz jLeftHip_rotz jLeftHip_rotx jRightBallFoot_roty jLeftKnee_roty jRightBallFoot_rotx jRightAnkle_rotz jRightAnkle_roty jRightKnee_rotx jLeftBallFoot_rotz jLeftHip_roty jRightHip_rotz jRightHip_rotx jLeftWrist_rotz jLeftElbow_roty jLeftElbow_rotx jRightBallFoot_rotz jC1Head_rotx jT1C7_roty jT9T8_rotx jLeftWrist_roty jT9T8_roty jT1C7_rotx jRightElbow_rotz jL1T12_rotz jL1T12_roty jL5S1_rotz jRightWrist_roty jT9T8_rotz jLeftElbow_rotz jL4L3_rotz jLeftShoulder_rotz jL1T12_rotx jL5S1_roty jL4L3_roty jL5S1_rotx jC1Head_rotz jRightShoulder_rotx jRightShoulder_roty jRightKnee_rotz jL4L3_rotx jRightShoulder_rotz jC1Head_roty jLeftC7Shoulder_rotz jRightC7Shoulder_rotx jRightC7Shoulder_roty jRightAnkle_rotx jRightC7Shoulder_rotz jRightKnee_roty jRightElbow_rotx jRightWrist_rotx jLeftKnee_rotx jT1C7_rotz jRightElbow_roty jRightWrist_rotz jLeftC7Shoulder_rotx jLeftWrist_rotx jLeftC7Shoulder_roty jRightHip_roty jLeftShoulder_rotx jLeftShoulder_roty) (-0.0329204631714089 -0.00766548540040273 -0.184990984986237 -0.0101714074323256 -0.0273673935794286 0.0521713298766476 0.0229562626639683 0.038555298042796 -0.0174015650102597 -0.0133621401807985 2.1640935598129e-05 0.0688944055088219 -0.111557381218557 -0.0211125869947517 -0.000474095965726264 0.158614992968627 -0.045322131733139 -0.0408436107210383 0.0284883352890699 0.247518932601934 0.0166177191111791 1.29660786536829e-06 0.0294155509540132 0.288509232083251 0.0029365080619979 -0.033597823866182 0.0247525773744096 0.0181191481091334 -0.00945956937667884 -0.0116557353851752 0.0329962494803435 -0.0263796521475483 -0.00208802510344413 -0.00873216577008965 -0.119108249236884 -0.0116599542984727 0.141513267541806 0.00395152282563002 0.0743232845748956 0.0339099666584104 0.00937444753612314 -0.0331573351900906 1.08388438631388 -0.182210978249635 -0.133178214525292 0.00396344848630914 0.0287412261875825 0.0166679068844259 -0.0403735264067343 0.0586253264076016 -0.0142663057869838 0.139429252787777 0.0537008433633771 -0.00654288940517647 0.0496537300267705 -0.0104596268877219 0.0109789426123595 -0.0198874072392417 0.0840406330821795 -0.0251233161332917 -0.0939721741803078 -0.029526158043161 -0.00576132109484614 0.200143967651348 -1.03996872886964 -0.400140738792913) (0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0) Pelvis -0.353565340690427 -2.64111424660093 0.929591201186423 0.751618666529599 0.0456384341958422 -0.0631250672903217 0.65498224352116 (0.0 0.0 0.0 0.0 0.0 0.0)

```
It follows that, any consumer device (e.g. `XsensRetargeting `, `HumanDynamicsEstimator`) is not simply reading the joint values and passing it to the model, but it is expected to use the joint values together to the joint name list.

The problem, in the case of @CarlottaSartore, arises to the fact that she is not using the interface of the `HumanStateProvider` (or the HumanStateProvider data streamed trough `human-state-wrapper`), but for her purpose she implemented a port that simply stream the joint values list as it is, without joint name information. 
Indeed this issue is, actually, just related to the implementation in [`wrappers/addVectorWisePublish`](https://github.com/robotology/human-dynamics-estimation/tree/wrappers/addVectorWisePublish).

lrapetti, 2020-11-26:
> why we offer instead a method to explicitly specify the joints you want to use and in which order using the `loadReducedModel`

This is what I was referring to in my [previous comment](https://github.com/robotology/human-dynamics-estimation/issues/222#issuecomment-734394424). In case we want to add a fixed-joint-order option, it would be better to use this functionality at the `HumanStateProvider` level, so that the whole flow of data (including the jointNameList) can be re-ordered.

traversaro, 2020-11-26:
> @S-Dafarra @traversaro probably my comment above was not clear.

Yes, sorry if I was not clear on my side, I just wanted to reinforce that you were already writing and for which you tagged me: the ModelLoader::loadModel order is completely arbitrary at the moment.

S-Dafarra, 2020-11-26:
> @S-Dafarra @traversaro probably my comment above was not clear.
> 
> We are aware that this joint order is not fixed, that's why we are not not trusting it at all. In fact, the output of `HumanStateProvider` is not a list of joint values, but a structured [interface exposing joint values and joint name list](https://github.com/robotology/human-dynamics-estimation/blob/master/msgs/yarp/thrift/HumanDynamicsEstimation/HumanState.thrift#L23-L34). This interface is then streamed on a yarp port using the `human-state-wrapper` device, this is how the output data looks:
> 
> ```
> -1 0.000000 2589.299452 (jLeftBallFoot_roty jLeftBallFoot_rotx jLeftAnkle_rotz jLeftAnkle_roty jLeftAnkle_rotx jLeftKnee_rotz jLeftHip_rotz jLeftHip_rotx jRightBallFoot_roty jLeftKnee_roty jRightBallFoot_rotx jRightAnkle_rotz jRightAnkle_roty jRightKnee_rotx jLeftBallFoot_rotz jLeftHip_roty jRightHip_rotz jRightHip_rotx jLeftWrist_rotz jLeftElbow_roty jLeftElbow_rotx jRightBallFoot_rotz jC1Head_rotx jT1C7_roty jT9T8_rotx jLeftWrist_roty jT9T8_roty jT1C7_rotx jRightElbow_rotz jL1T12_rotz jL1T12_roty jL5S1_rotz jRightWrist_roty jT9T8_rotz jLeftElbow_rotz jL4L3_rotz jLeftShoulder_rotz jL1T12_rotx jL5S1_roty jL4L3_roty jL5S1_rotx jC1Head_rotz jRightShoulder_rotx jRightShoulder_roty jRightKnee_rotz jL4L3_rotx jRightShoulder_rotz jC1Head_roty jLeftC7Shoulder_rotz jRightC7Shoulder_rotx jRightC7Shoulder_roty jRightAnkle_rotx jRightC7Shoulder_rotz jRightKnee_roty jRightElbow_rotx jRightWrist_rotx jLeftKnee_rotx jT1C7_rotz jRightElbow_roty jRightWrist_rotz jLeftC7Shoulder_rotx jLeftWrist_rotx jLeftC7Shoulder_roty jRightHip_roty jLeftShoulder_rotx jLeftShoulder_roty) (-0.0329204631714089 -0.00766548540040273 -0.184990984986237 -0.0101714074323256 -0.0273673935794286 0.0521713298766476 0.0229562626639683 0.038555298042796 -0.0174015650102597 -0.0133621401807985 2.1640935598129e-05 0.0688944055088219 -0.111557381218557 -0.0211125869947517 -0.000474095965726264 0.158614992968627 -0.045322131733139 -0.0408436107210383 0.0284883352890699 0.247518932601934 0.0166177191111791 1.29660786536829e-06 0.0294155509540132 0.288509232083251 0.0029365080619979 -0.033597823866182 0.0247525773744096 0.0181191481091334 -0.00945956937667884 -0.0116557353851752 0.0329962494803435 -0.0263796521475483 -0.00208802510344413 -0.00873216577008965 -0.119108249236884 -0.0116599542984727 0.141513267541806 0.00395152282563002 0.0743232845748956 0.0339099666584104 0.00937444753612314 -0.0331573351900906 1.08388438631388 -0.182210978249635 -0.133178214525292 0.00396344848630914 0.0287412261875825 0.0166679068844259 -0.0403735264067343 0.0586253264076016 -0.0142663057869838 0.139429252787777 0.0537008433633771 -0.00654288940517647 0.0496537300267705 -0.0104596268877219 0.0109789426123595 -0.0198874072392417 0.0840406330821795 -0.0251233161332917 -0.0939721741803078 -0.029526158043161 -0.00576132109484614 0.200143967651348 -1.03996872886964 -0.400140738792913) (0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0) Pelvis -0.353565340690427 -2.64111424660093 0.929591201186423 0.751618666529599 0.0456384341958422 -0.0631250672903217 0.65498224352116 (0.0 0.0 0.0 0.0 0.0 0.0)
> ```
> 
> It follows that, any consumer device (e.g. `XsensRetargeting `, `HumanDynamicsEstimator`) is not simply reading the joint values and passing it to the model, but it is expected to use the joint values together to the joint name list.
> 
> The problem, in the case of @CarlottaSartore, arises to the fact that she is not using the interface of the `HumanStateProvider` (or the data streamed trough `human-state-wrapper`), but for her purpose she implemented a port that simply stream the joint values list as it is, without joint name information.
> Indeed this issue is, actually, just related to the implementation in [`wrappers/addVectorWisePublish`](https://github.com/robotology/human-dynamics-estimation/tree/wrappers/addVectorWisePublish).

This was indeed very clear. I understood that this problem was only relative to [`wrappers/addVectorWisePublish`](https://github.com/robotology/human-dynamics-estimation/tree/wrappers/addVectorWisePublish). That new port is the one I am expecting to throw the lottery numbers.

At his point, I agree with any solution that allow removing this randomness. It also makes sense to not prescribe any order at all,  if through the interface the order is clear.

The open point then, is to determine the joint values order in a port that streams only doubles. 

> In case we want to add a fixed-joint-order option, it would be better to use this functionality at the `HumanStateProvider` level, so that the whole flow of data (including the jointNameList) is re-ordered.

I see. The point, is how to set this order. Configuration files are tricky, as you are relying on the person, rather than the application. What I was proposing was to use another port to determine this ordering. A sort of equivalent of API call, but from a different process using another language.

CarlottaSartore, 2020-11-26:
> At this point, I would rather expect the port that outputs only the joint values not to send anything, unless a specific order is first sent to it

The current implementation I am using right now orders the joints by reading a parameter, if that parameter is not present it uses the same order as the one published by the human-state-provider (i.e. the one returned by the iDynTree::loadReducedModelFromFile). 

If I get correctly what you want to say @S-Dafarra  is that instead, if no order is stated, the output should be empty, am I right? 

> The advantage is avoiding hard-coded joint order lists in the HumanStateProvider and in the "consumer device" (whoever is reading the joint values), that would need to be maintained together

@lrapetti  regarding this, we could implement the possibility to either give the order or not, in this way it could be the user choice to output the joints in a certain order, hence not breaking compatibility with other things you developed. 

>  So feel free to open a PR to add the feature you have implemented.

About this, I think first we should agree on how and where implement this feature,

> Just a small comment on the implementation, I was thinking that, instead of human-state-wrapper, this feature might be implemented directly in the HumanStateProvider using loadReducedModelFromFile [...]

it seems to me that this is the best choice, we should decide also  what to do if the joints order is not specified e.g.: 
- Publishing nothing; (as @S-Dafarra  suggested if I correctly understood); 
- Publishing the order given by iDynTree

@S-Dafarra I didn't get this point can you please clarify it to me? Thanks 

> What I was proposing was to use another port to determine this ordering. A sort of equivalent to an API call, but from a different process using another language.