gbionics/human-dynamics-estimation #231: Release v2.2.0

lrapetti, 2021-01-29:
The last release was done almost one year ago with the [`v.1.0` tag](https://github.com/robotology/human-dynamics-estimation/releases/tag/v2.1.0), and since then new features have been released. I think it's time to release it.

Indeed we should proceed:
- [ ] merging `devel` into `master`
- [ ] creating the new release

I think also we might align with the rest of `robotology` removing the `devel` branch, and keeping the released tags as stable code pointers, leaving `master` as development branch.

cc @Yeshasvitvs @traversaro @diegoferigo

Yeshasvitvs, 2021-01-29:
@lrapetti I agree with it.

I hoped to had [test/devel-rebase](https://github.com/Yeshasvitvs/human-dynamics-estimation/commits/test/devel-rebase) tested before making a new release, as this would have enabled to pull in the changes from https://github.com/Yeshasvitvs/human-dynamics-estimation/commits/test/devel-rebase to upstream.

However, at this point it is cleaner have a release from merging devel to master.

> I think also we might align with the rest of robotology removing the devel branch, keeping the released tags as stable code pointers, and using master as development branch.

Good call. This will make things aligned with how [wearable](https://github.com/robotology/wearables) is maintained too.

Yeshasvitvs, 2021-01-29:
Before the release, may be checking the config yarp manager files for iCub visualization through https://github.com/robotology/human-dynamics-estimation/pull/230 can be useful to have tested and committed to upstream for the next iteration of tests for retargeting.

@S-Dafarra

Yeshasvitvs, 2021-01-29:
@lrapetti  merged `devel` into `master` branch https://github.com/robotology/human-dynamics-estimation/commit/d6fdd7a8627ef71dc49225d716e11aff7ab043bc

lrapetti, 2021-01-29:
> I hoped to had [test/devel-rebase](https://github.com/Yeshasvitvs/human-dynamics-estimation/commits/test/devel-rebase) tested before making a new release, as this would have enabled to pull in the changes from #181 to upstream.

The main problem there is that it depends on a branch of `iDynTree` (https://github.com/robotology/idyntree/pull/783), so for the time being that branch can not be merged to a stable release.

Yeshasvitvs, 2021-01-29:
Windows CI failed on push flow after merge at https://github.com/robotology/human-dynamics-estimation/runs/1790878536#step:9:10886, looking for `IRRLICHT_INCLUDE_DIR`

lrapetti, 2021-02-01:
> Windows CI failed on push flow after merge at https://github.com/robotology/human-dynamics-estimation/runs/1790878536#step:9:10886, looking for `IRRLICHT_INCLUDE_DIR`

The failure has now been fixed (https://github.com/robotology/human-dynamics-estimation/pull/232), @Yeshasvitvs we can create the new release, what do you think?

Yeshasvitvs, 2021-02-03:
@lrapetti following https://github.com/robotology/human-dynamics-estimation/pull/235 https://github.com/robotology/human-dynamics-estimation/pull/234 https://github.com/robotology/human-dynamics-estimation/pull/233 
do you foresee any further small changes that are to be done before a release ?

lrapetti, 2021-02-03:
> do you foresee any further small changes that are to be done before a release ?

No, I am not planning any change, for me we can proceed with the release.

Yeshasvitvs, 2021-02-03:
@lrapetti and I did the new release https://github.com/robotology/human-dynamics-estimation/releases/tag/v2.2.0

Yeshasvitvs, 2021-02-03:
Expecting an automatic update of [robotology-superbuild](https://github.com/robotology/robotology-superbuild), similar to https://github.com/robotology/robotology-superbuild/pull/607

Yeshasvitvs, 2021-02-03:
> I think also we might align with the rest of robotology removing the devel branch, keeping the released tags as stable code pointers, and using master as development branch.

@lrapetti We can open separate issue for this task, and also align https://github.com/robotology/wearables with release tag branches

closed 2021-02-03 by Yeshasvitvs

traversaro, 2021-02-03:
> I think also we might align with the rest of robotology removing the devel branch, keeping the released tags as stable code pointers, and using master as development branch.

Just to clarify, not using devel is a strategy that make sense and I think it make sense to use for you, but it is not used by all robotology projects. At the moment is used by YARP and YCM, but not (for example) by icub-main . At the moment the `master`  branch of this repo is still the default branch used by anyone that downloads the robotology-superbuild and compiles it with the default settings with `ROBOTOLOGY_ENABLE_HUMAN_DYNAMICS` , that is ROBOTOLOGY_PROJECT_TAGS set to Stable. In the future we may change this default to use the latest-releases.yaml file introduced in https://github.com/robotology/robotology-superbuild/pull/596 , but probably we will need to discuss about it before doing so.

lrapetti, 2021-02-03:
Cross linking the comment from @traversaro https://github.com/robotology/human-dynamics-estimation/pull/233#issuecomment-772591190. 

In brief this release is currently not aligned with the stable branch of iDynTree if `Irrlicht` is enabled. For the same reason, the CI is currenty pointing to the iDynTree `devel` branch (https://github.com/robotology/human-dynamics-estimation/blob/master/.github/workflows/ci.yml#L19), this will change as iDynTree will be relased (https://github.com/robotology/idyntree/issues/804).

traversaro, 2021-02-03:
> 
> 
> Expecting an automatic update of [robotology-superbuild](https://github.com/robotology/robotology-superbuild), similar to [robotology/robotology-superbuild#607](https://github.com/robotology/robotology-superbuild/pull/607)

FYI that scripts runs every sunday night (see https://github.com/robotology/robotology-superbuild/blob/8f415ed824730de37d886806e1db58d88117a6be/.github/workflows/update-latest-releases.yml#L9), so if you want to bump the release before manually feel free to do so.