gbionics/human-dynamics-estimation #250: Move berdy into hde library

lrapetti, 2021-05-25:
As discussed in https://github.com/robotology/human-dynamics-estimation/issues/243, we should proceed to move the berdy algorithm implementation [out of `iDynTree`](https://github.com/robotology/idyntree/blob/master/src/estimation/include/iDynTree/Estimation/BerdySparseMAPSolver.h) into hde library.

From iDynTree side, I think it was discussed with @traversaro and there should be no problem.

We can proceed by moving berdy, as it is, into the library changing all the related namespaces. Once that is done, and the hde master works fine, we can update berdy with the sot implementation (from https://github.com/robotology/idyntree/pull/783)

lrapetti, 2021-05-26:
> > From iDynTree side, I think it was discussed with @traversaro and there should be no problem.
> 
> Sure, the only problem is if you want to have matlab bindings in HDE (that is doable, but not trivial).

Sorry, I forgot this point. Anyway, probably sooner o later we should address it.

Anyway, this is relevant for @claudia-lat.

cc @DanielePucci

traversaro, 2021-05-26:
A possible way forward inline with recent devs in the lab is to wrap the code in Python with pybind11 or SWIG, and then call the resulting Python from MATLAB (see https://github.com/dic-iit/bipedal-locomotion-framework/issues/53#issuecomment-789058776).

lrapetti, 2021-06-07:
I did an initial test trying to port the current berdy implementation (from iDynTree master) in the branch [`human-dynamics-estimation/tree/add-berdy`](https://github.com/robotology/human-dynamics-estimation/tree/add-berdy). Everything seems to be compiling fine in Ubuntu, but I am getting some compilation error in MacOS and Windows CI.

I will try to investigate further. 

If those errors are fixed, I think the porting would be done. However, I think before moving definitely to hde-berdy, we should take care of the problem of the bindings.

> A possible way forward inline with recent devs in the lab is to wrap the code in Python with pybind11 or SWIG, and then call the resulting Python from MATLAB (see [dic-iit/bipedal-locomotion-framework#53 (comment)](https://github.com/dic-iit/bipedal-locomotion-framework/issues/53#issuecomment-789058776)).

I have opened a specific issue for this https://github.com/robotology/human-dynamics-estimation/issues/255.

lrapetti, 2021-06-07:
> Everything seems to be compiling fine in Ubuntu, but I am getting some compilation error in MacOS and Windows CI.

in particular the problem is due to the lines that are uncommented in https://github.com/robotology/human-dynamics-estimation/commit/46951de55d67f3e93b4a0d7c6a580ab8bd209792. Without those lines, I was able to compile and run everything on `MacOS`, and all the CI tests were passed.