gbionics/human-dynamics-estimation #253: New release of human-dynamics estimation

lrapetti, 2021-05-27:
I think that since we merged the PR related to the hde library (https://github.com/robotology/human-dynamics-estimation/pull/246), and the installation is fixed and merged (https://github.com/robotology/human-dynamics-estimation/pull/251), we can do a new release tag for `human-dynamics-estimation` `v2.3.0`

Yeshasvitvs, 2021-05-27:
@lrapetti we can wait for this week to ensure we don't encounter any other problems during teleoperation tests, and then proceed with the new release

CC @kouroshD may be we can also co-ordinate in making a new release of https://github.com/robotology/walking-teleoperation following the tests

lrapetti, 2021-08-03:
Cross linking @traversaro comment https://github.com/robotology/walking-teleoperation/issues/65#issuecomment-891823461. @Yeshasvitvs if you are ok I can proceed with the new tag release for `human-dynamics-estimation`, while I think @kouroshD will handle the release in `walking-teleoperation`

Yeshasvitvs, 2021-08-03:
Please go ahead with the release @lrapetti 
Thank you!

closed 2021-08-23 by lrapetti