gbionics/human-dynamics-estimation #259: Fix HumanStateProvider on Windows

lrapetti, 2021-07-26:
We have tried to run the `HumanStateProvider` and the visualizer on a Windows machine, and we have noticed a strange behaviour.

Results are shown in the following video where we run the HSP+visualization on the same dataset and same configuration ([`Human.xml`](https://github.com/robotology/human-dynamics-estimation/blob/master/conf/xml/Human.xml)):

https://user-images.githubusercontent.com/35487806/127000120-ce887aa1-903b-4a67-b870-f166a77cfe36.mp4

There are two main problems:
 - (A) IK output seems to be wrong
 - (B) When we stop the data, we expect the IK output to be fixed, instead it keeps changing.

Some possibl source of problem could be:
- Visualizer (it seems not since it wouldn't explain problem (B)). But we can easily double-check by running HSP on Windows and visualizer on another machine
- Some non initialized matrices. In this case running in debug the problem shouldn't happen
- IK parameters tuning

S-Dafarra, 2021-07-26:
> * Visualizer (it seems not since it wouldn't explain problem (B)). But we can easily double-check by running HSP on Windows and visualizer on another machine

The reverse (Visualizer on Windows and HSP on another machine) could be interesting too.

traversaro, 2021-07-26:
Is this using vcpkg or conda dependencies? How is the locale set in the different machines?

lrapetti, 2021-07-27:
> Is this using vcpkg or conda dependencies? How is the locale set in the different machines?

On top, I am on MacOS, dependencies installed via `homebrew` and `human-state-provider` installed via `robotology-superbuild`. Same behaviour is experienced on other machines on Ubuntu (`apt`+`superbuild`).

On the Windows machine (visualization on the bottom), dependencies are installed via `vcpkg` and `human-state-provider` via `robotology-superbuild`. (please @kouroshD correct me if I am wrong)

traversaro, 2021-07-27:
> Same behaviour is experienced on other machines on Ubuntu (`apt`+`superbuild`).

The same problem of this issue?

lrapetti, 2021-07-27:
> > Same behaviour is experienced on other machines on Ubuntu (`apt`+`superbuild`).
> 
> The same behaviour described in this issue?

No, on Ubuntu we have the same behaviour that we experience on MacOS.
The problem appears to be only on Windows.

traversaro, 2021-07-27:
If the functionality is part of a released version of `human-dynamics-estimation`, a quick check you can do without recompiling the world is to just install `human-dynamics-estimation` via conda binaries on Windows:
~~~
mamba create -n hde-env
mamba activate hde-env
mamba install -c conda-forge -c robotology human-dynamics-estimation
# Make sure that all environment variables are properly set
mamba deactivate hde-env
mamba activate hde-env
~~~

If you are already in machine where the robotology-superbuild with vcpkg is installed, make sure that you clean the enviornment from any globally set env variables, using the appropriate powershell scripts.

kouroshD, 2021-07-27:
>  How is the locale set in the different machines?

@traversaro  what do you mean by `locale` here? Is it related to https://docs.microsoft.com/en-us/windows/win32/intl/locales-and-languages ?

traversaro, 2021-07-27:
> 
> 
> > How is the locale set in the different machines?
> 
> @traversaro what do you mean by `locale` here? Is it related to https://docs.microsoft.com/en-us/windows/win32/intl/locales-and-languages ?

Yes, kind of. I asked that as there are a lot of bugs related to the use of functions that convert string to floating point numbers (and back and forth). See https://github.com/robotology/idyntree/issues/288 . See https://serverfault.com/questions/173630/how-to-get-system-locale-in-windows-7-cmd for how to get it.

lrapetti, 2021-07-28:
> * But we can easily double-check by running HSP on Windows and visualizer on another machine

We did this test and visualization still seems to be wrong, so we can exclude it is a problem of the visualizer.
We also noticed that the `human-state-provider` on Windows, from time to time was starting printing the warning for velocity out of the limit, with huge powers.

lrapetti, 2023-03-22:
This problem has not been reported for a while, and I know that the `HumanStateProvider` was running on windows on @LorenzoFiori and @AntonellaTatarelli laptop.

I will close this issue, we will re-open it in case we experience the same problem again.

closed 2023-03-22 by lrapetti