gbionics/human-dynamics-estimation #284: Gravity target not working properly

lrapetti, 2022-02-24:
With @S-Dafarra we have observed that the gravity target introduced in https://github.com/robotology/human-dynamics-estimation/pull/281, in the `add-different-target-types` branch is not working properly.

As you can see in the following video, the target gravity vector orientation and the current gravity vector orientation (the two orange arrow in correspondence of the links) seems that certain configuration do not get aligned, but rather they diverge (this visualization has been obtained by modifying the current visualizer [`HumanStateVisualizer`](https://github.com/robotology/human-dynamics-estimation/tree/add-different-target-types/modules/HumanStateVisualizer)). Those two vectors are exactly the vectors for which we compute the error in https://github.com/robotology/human-dynamics-estimation/blob/e19082754529f2355f437c8e71729df5611a8452/HumanDynamicsEstimationLibrary/algorithms/src/DynamicalInverseKinematics.cpp#L569
corresponding to the following formula 
$$
\omega = -k (z^{target} \times z^{current}) + \omega^{target}
$$

https://user-images.githubusercontent.com/35487806/155574488-b5a8609a-8737-492b-b394-a9301bb46eb9.mp4

lrapetti, 2022-02-24:
As noticed by @S-Dafarra, in the formula above the cross product give us the desired angular velocity in the link frame. Indeed, in order to use it in the implemented Inverse Kinematics (see https://www.mdpi.com/1999-4893/13/10/266) we would need that quantity expressed in the world inertial frame, i.e.
$$
_{I}\omega = -k _IR^{B}(z^{target} \times z^{current}) + \omega^{target}
$$

lrapetti, 2022-02-25:
Fix merged with https://github.com/robotology/human-dynamics-estimation/pull/285.

closed 2022-02-25 by lrapetti

DanielePucci, 2022-02-26:
Nice catch @lrapetti @S-Dafarra