gbionics/human-dynamics-estimation #294: Implement a HumanDynamicsRemapper

Zweisteine96, 2022-04-14:
Hi @lrapetti in this issue we want to implement a HumanDynamicsRemapper.

lrapetti, 2022-04-15:
Just to add a bit of extra information, the role of the Remapper YARP device in general is to expose the data coming from a thrifted yarp data port, trought a cpp interface.
The `HumanDynamicsRemapper` will:
- **read** [HumanDynamics.thrift](https://github.com/robotology/human-dynamics-estimation/blob/master/msgs/yarp/thrift/hde/msgs/HumanDynamics.thrift) message
- **expose** [IHumanDynamics](https://github.com/robotology/human-dynamics-estimation/blob/master/interfaces/IHumanDynamics/hde/interfaces/IHumanDynamics.h) interface.

As an example, from which you can start, it's the [`HumanStateRemapper`](https://github.com/robotology/human-dynamics-estimation/tree/master/remappers/HumanStateRemapper) device which reads [`HumanState.thrift`](https://github.com/robotology/human-dynamics-estimation/blob/master/msgs/yarp/thrift/hde/msgs/HumanState.thrift) messages, exposing the [`IHumanState`](https://github.com/robotology/human-dynamics-estimation/tree/master/interfaces/IHumanState/hde/interfaces) interface.
As you can observe, looking at the `HumanStateRemapper` class header you can observe that:
- inherit from the interface that it wants to expose (see [here](https://github.com/robotology/human-dynamics-estimation/blob/master/remappers/HumanStateRemapper/HumanStateRemapper.h#L30))
- implement the callback to read the thrifted message (in this case `hde::msgs::HumanState`) (see [here](https://github.com/robotology/human-dynamics-estimation/blob/80b5e67dde7a8f9b8e675248ff3ad9109181ae7f/remappers/HumanStateRemapper/HumanStateRemapper.h#L31))
- implement the periodic thread (see [here](https://github.com/robotology/human-dynamics-estimation/blob/80b5e67dde7a8f9b8e675248ff3ad9109181ae7f/remappers/HumanStateRemapper/HumanStateRemapper.h#L32)). (⚠️ it seems that the [run()](https://github.com/robotology/human-dynamics-estimation/blob/80b5e67dde7a8f9b8e675248ff3ad9109181ae7f/remappers/HumanStateRemapper/HumanStateRemapper.cpp#L131-L134) method is currently empty, so I as far as I see it is not needed to have the periodic thread)

Then in the device implementation ([`HumanStateRemapper.cpp`](https://github.com/robotology/human-dynamics-estimation/blob/master/remappers/HumanStateRemapper/HumanStateRemapper.cpp)), the relevant part are:
- the open method where the configuration file is parsed and the connections are done (see [here](https://github.com/robotology/human-dynamics-estimation/blob/80b5e67dde7a8f9b8e675248ff3ad9109181ae7f/remappers/HumanStateRemapper/HumanStateRemapper.cpp#L60-L115))
- the [`onRead`](https://github.com/robotology/human-dynamics-estimation/blob/80b5e67dde7a8f9b8e675248ff3ad9109181ae7f/remappers/HumanStateRemapper/HumanStateRemapper.cpp#L136-L154) method where data are read from the port and saved in buffer variables
- the methods of the `IHumanState` interface are exposing the buffer variables data (see [here](https://github.com/robotology/human-dynamics-estimation/blob/80b5e67dde7a8f9b8e675248ff3ad9109181ae7f/remappers/HumanStateRemapper/HumanStateRemapper.cpp#L156-L203))

cc @RiccardoGrieco

Zweisteine96, 2022-04-18:
Hi @lrapetti, I updated the following files to[ this branch ](https://github.com/robotology/human-dynamics-estimation/tree/humanDynamicsRemapper/remappers)in HDE repo:
- `humanDynamicsRemapper.h`
- `humanDynamicsRemapper.cpp`
- `CMakeLists.txt`

About the humanDynamicsRemapper, there remains one point that I'm not very sure, that is, how could I determine the `humanDynamicsDataPort` shown here: https://github.com/robotology/human-dynamics-estimation/blob/humanDynamicsRemapper/remappers/HumanDynamicsRemapper/HumanDynamicsRemapper.cpp#L59.

lrapetti, 2022-04-19:
> About the humanDynamicsRemapper, there remains one point that I'm not very sure, that is, how could I determine the `humanDynamicsDataPort` shown here: https://github.com/robotology/human-dynamics-estimation/blob/humanDynamicsRemapper/remappers/HumanDynamicsRemapper/HumanDynamicsRemapper.cpp#L59.

This parameter will be passed trough configuration file as it is done for `HumanStateRemapper` in https://github.com/robotology/human-dynamics-estimation/blob/80b5e67dde7a8f9b8e675248ff3ad9109181ae7f/conf/xml/RobotPositionController_iCub.xml#L6-L8

Zweisteine96, 2022-04-19:
I checked the configuration file here: https://github.com/robotology/human-dynamics-estimation/tree/80b5e67dde7a8f9b8e675248ff3ad9109181ae7f/conf/xml, but didn't find one that is related to the human_dynamics_remapper, did I miss something? Or should I add one such configuration file manually?

lrapetti, 2022-04-19:
> didn't find one that is related to the human_dynamics_remapper, did I miss something? Or should I add one such configuration file manually?

Since it was not implemented, there are no configuration files for `human_dynamics_remapper` so you should create one.

What you can test is to try to run a yarprobotinterface with the `human_dynamics_wrapper` such as the file `Human.xml` (it has the dynamics wrapper here https://github.com/robotology/human-dynamics-estimation/blob/80b5e67dde7a8f9b8e675248ff3ad9109181ae7f/conf/xml/Human.xml#L223-L232).

And then create a new `yarprobotinterface` with the `human_dynamics_wrapper` only and run it and see if it runs fine.

Zweisteine96, 2022-04-27:
The humanDynamicsRemapper has already been implemented and an [PR](https://github.com/robotology/human-dynamics-estimation/pull/295#pullrequestreview-954720034) is opened for this modification. The naming problem will be fixed in the future. Now close this issue.

lrapetti, 2022-04-27:
> The humanDynamicsRemapper has already been implemented and an [PR](https://github.com/robotology/human-dynamics-estimation/pull/295#pullrequestreview-954720034) is opened for this modification. The naming problem will be fixed in the future. Now close this issue.

I am reopening the issue since usually we close issues after merging the associated PR.

lrapetti, 2022-05-02:
Now the PR is merged, we can close it

closed 2022-05-02 by lrapetti