gbionics/human-dynamics-estimation #304: Strange tag name for 2.5.0 release

traversaro, 2022-05-30:
Just a nitpick/note: all the other releases tag started with a lowercase `v`:
* https://github.com/robotology/human-dynamics-estimation/releases/tag/v2.3.0
* https://github.com/robotology/human-dynamics-estimation/releases/tag/v2.4.0
* https://github.com/robotology/human-dynamics-estimation/releases/tag/v2.4.1

2.5.0 switched to a capital `V`, not sure if this was intentional:
* https://github.com/robotology/human-dynamics-estimation/releases/tag/V2.5.0

fyi @lrapetti

lrapetti, 2022-05-30:
I have deleted and created again the release, now it can be found under the tag `v2.5.0` https://github.com/robotology/human-dynamics-estimation/releases/tag/v2.5.0.

Thanks for noticing it @traversaro

traversaro, 2022-05-30:
Thanks for fixing it

closed 2022-05-30 by traversaro