gbionics/human-dynamics-estimation #311: Migrate Rviz support from ROS1 to ROS2

traversaro, 2022-08-07:
Upstream effort from the ROS mantainers has shifted from years from ROS1 to ROS2. To avoid future mantainance problem, it would be great to migrate the part of this repo that depend on ROS1 to ROS2 (fortunatly it should not be a lot).

A generic documentation for this is available in https://docs.ros.org/en/humble/The-ROS2-Project/Contributing/Migration-Guide.html, but for this specific task the script proposed in https://github.com/ros2/rviz/pull/882 could be quite useful.

DanielePucci, 2022-08-20:
CC @lrapetti @RiccardoGrieco

traversaro, 2023-01-25:
It would be great if this could be inserted in the next Research Increment planning as part of AMI lab-wide effort to "explor" use of ROS2 .

fyi @lrapetti @RiccardoGrieco @DanielePucci @CarlottaSartore

traversaro, 2023-01-26:
> the script proposed in [ros2/rviz#882](https://github.com/ros2/rviz/pull/882) could be quite useful.

If that script creates problem, we could also make sure that we are using the latest version with the fixes proposed in https://github.com/ros2/rviz/pull/945 .

lrapetti, 2023-02-01:
Hi, I managed to install ros2 on mac using `conda` (following instructions in https://robostack.github.io/GettingStarted.html) and it seems that is working (e.g. running `rviz2`).

Then I tried to run the script `rviz1_to_rviz2.py` suggested by @traversaro, but it is not working:
```
$ ros2 run rviz2 rviz1_to_rviz2.py
No executable found
```
I am not very familiar with ros2, and I don't know how/if the python script should be found

traversaro, 2023-02-01:
My bad, I just realized that the script is only available on rolling (i.e. the experimental version) and it is still not on humble (the LTS version). Anyhow, as it is a simple Python script, you can just download it and run it, i.e. : 
~~~
wget https://raw.githubusercontent.com/ros2/rviz/rolling/rviz2/scripts/rviz1_to_rviz2.py
python rviz1_to_rviz2.py
~~~

Anyhow, I probably realized I was a bit too optimistic. Once we got rviz2 to visualize the human, we also need to migrate the appropriate publisher devices in https://github.com/robotology/human-dynamics-estimation/tree/d3a2f546a651ffb17e17b9b7e1b4933abe7e4c3d/publishers from ROS1 to ROS2 . Differently from ROS1, there is no direct support in the YARP library for publishing ROS2 messages, so we need to publish on ROS2 linking directly ROS2 libraries. Some good examples of YARP devices that publish ROS2 messages can be found inhttps://github.com/robotology-playground/yarp-devices-ros2 .

lrapetti, 2023-02-02:
> My bad, I just realized that the script is only available on rolling (i.e. the experimental version) and it is still not on humble (the LTS version). Anyhow, as it is a simple Python script, you can [...]

Concerning this I tried to use the script. At first I got the following error:
```
$python rviz1_to_rviz2.py robotology-superbuild/src/HumanDynamicsEstimation/conf/ros/rviz/HDERviz.rviz robotology-superbuild/src/HumanDynamicsEstimation/conf/ros/rviz/HDERviz2.rviz
Traceback (most recent call last):
  File "/Users/lorenzorapetti/Software/rviz1_to_rviz2.py", line 492, in <module>
    main()
  File "/Users/lorenzorapetti/Software/rviz1_to_rviz2.py", line 487, in main
    rviz2_config = migrate_to_rviz2(rviz1_config)
  File "/Users/lorenzorapetti/Software/rviz1_to_rviz2.py", line 453, in migrate_to_rviz2
    rviz2_config[vm_key] = migrate_visualization_manager(rviz1_config[vm_key])
  File "/Users/lorenzorapetti/Software/rviz1_to_rviz2.py", line 77, in migrate_visualization_manager
    'Displays': migrate_displays(vm_dict['Displays']),
  File "/Users/lorenzorapetti/Software/rviz1_to_rviz2.py", line 128, in migrate_displays
    displays_rviz2.append(migration_functions[display_dict['Class']](display_dict))
  File "/Users/lorenzorapetti/Software/rviz1_to_rviz2.py", line 288, in migrate_display_wrench_stamped
    depth=display_dict['Queue Size'],
KeyError: 'Queue Size'
```

Then I tried to modify the original file `HDERviz2.rviz` but adding the `Queue Size` option to the wrench stamped classes, 
e.g.:
```
- Alpha: 1
      Arrow Width: 0.30000001192092896
      Class: rviz/WrenchStamped
      Enabled: true
      Force Arrow Scale: 0.0020000000949949026
      Force Color: 255; 239; 6
      Hide Small Values: true
      History Length: 1
      Name: LeftFoot_ExtWrenches
      Topic: /HDE/WrenchStamped/LeftFoot
      Torque Arrow Scale: 0
      Torque Color: 37; 95; 255
      Unreliable: false
      Queue Size: 10
      Value: true
```

And it seems to work, here is the loaded rviz2 configuration
[image]

The modified and the new rviz configuration files have been uploaded in https://github.com/robotology/human-dynamics-estimation/tree/feature/rviz2.

lrapetti, 2023-02-02:
I just noticed that the following warning was raised when running the `rviz2` file:
```
$rviz2 -d HDERviz2.rviz 
[INFO] [1675329620.117061682] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1675329620.117206106] [rviz2]: OpenGl version: 2.1 (GLSL 1.2)
[INFO] [1675329620.167655129] [rviz2]: Stereo is NOT SUPPORTED
Warning: Invalid frame ID "ground" passed to canTransform argument target_frame - frame does not exist
         at line 93 in /Users/runner/mambaforge/conda-bld/ros-humble-tf2_1660973751483/work/ros-humble-tf2/src/work/src/buffer_core.cpp
Warning: Invalid frame ID "ground" passed to canTransform argument target_frame - frame does not exist
         at line 93 in /Users/runner/mambaforge/conda-bld/ros-humble-tf2_1660973751483/work/ros-humble-tf2/src/work/src/buffer_core.cpp
Warning: Invalid frame ID "ground" passed to canTransform argument target_frame - frame does not exist
         at line 93 in /Users/runner/mambaforge/conda-bld/ros-humble-tf2_1660973751483/work/ros-humble-tf2/src/work/src/buffer_core.cpp
Warning: Invalid frame ID "ground" passed to canTransform argument target_frame - frame does not exist
         at line 93 in /Users/runner/mambaforge/conda-bld/ros-humble-tf2_1660973751483/work/ros-humble-tf2/src/work/src/buffer_core.cpp
Warning: Invalid frame ID "ground" passed to canTransform argument target_frame - frame does not exist
         at line 93 in /Users/runner/mambaforge/conda-bld/ros-humble-tf2_1660973751483/work/ros-humble-tf2/src/work/src/buffer_core.cpp
Warning: Invalid frame ID "ground" passed to canTransform argument target_frame - frame does not exist
         at line 93 in /Users/runner/mambaforge/conda-bld/ros-humble-tf2_1660973751483/work/ros-humble-tf2/src/work/src/buffer_core.cpp
Warning: Invalid frame ID "ground" passed to canTransform argument target_frame - frame does not exist
         at line 93 in /Users/runner/mambaforge/conda-bld/ros-humble-tf2_1660973751483/work/ros-humble-tf2/src/work/src/buffer_core.cpp
Warning: Invalid frame ID "ground" passed to canTransform argument target_frame - frame does not exist
         at line 93 in /Users/runner/mambaforge/conda-bld/ros-humble-tf2_1660973751483/work/ros-humble-tf2/src/work/src/buffer_core.cpp
Warning: Invalid frame ID "ground" passed to canTransform argument target_frame - frame does not exist
         at line 93 in /Users/runner/mambaforge/conda-bld/ros-humble-tf2_1660973751483/work/ros-humble-tf2/src/work/src/buffer_core.cpp
Warning: Invalid frame ID "ground" passed to canTransform argument target_frame - frame does not exist
         at line 93 in /Users/runner/mambaforge/conda-bld/ros-humble-tf2_1660973751483/work/ros-humble-tf2/src/work/src/buffer_core.cpp
Warning: Invalid frame ID "ground" passed to canTransform argument target_frame - frame does not exist
         at line 93 in /Users/runner/mambaforge/conda-bld/ros-humble-tf2_1660973751483/work/ros-humble-tf2/src/work/src/buffer_core.cpp
Warning: Invalid frame ID "ground" passed to canTransform argument target_frame - frame does not exist
...
```

to be investigated

traversaro, 2023-02-02:
Probably the warning is because no tf messages are published?

traversaro, 2023-02-02:
> 
> Yess, I think might be a good exercise to do anyway. On this I also have another question, we are currently using the `yarprobotstatepublisher` to publish the transforms of the links used in rviz. Is there an equivalent working for `ros2`?

`yarprobotstatepublisher` is reading data from the `/jointState` topic or other sources, and publishing it via the [`transformClient`](https://www.yarp.it/latest/classTransformClient.html) YARP device. In general, the `transformClient` in YARP is deprecated, and can be replaced by the devices described in https://www.yarp.it/latest/group__FrameTransform.html and https://www.yarp.it/latest/group__Fts__ftc__config.html . When using this new devices, it is possible to publish the TFs to ROS2 via the [`frameTransformSet_nwc_ros2`](https://github.com/robotology-playground/yarp-devices-ros2/tree/master/src/devices/frameTransformSet_nwc_ros2) from the `yarp-devices-ros2` . However, this may requires changes to `yarprobotstatepublisher` that may not be trivial. 

An alternative is to just publish a ROS2 `/jointState` topic, and then generate the corresponding TFs via the classical ROS2 node `robotstatepublisher`, see https://docs.ros.org/en/humble/Tutorials/Intermediate/URDF/Using-URDF-with-Robot-State-Publisher.html#publish-the-state . Note that you can use the robotstatepublisher simply via `ros2 run robot_state_publisher robot_state_publisher`, there is no requirement for using the relatively complex launch files via python as shown in the tutorial.

lrapetti, 2023-02-07:
The discussion on ROS2 publishers is moved to https://github.com/robotology/human-dynamics-estimation/issues/334