gbionics/human-dynamics-estimation #318: human_wrench_provider returning zeros if not attached to HumanStateProvider

lrapetti, 2022-09-23:
After merging https://github.com/robotology/human-dynamics-estimation/pull/309, the human_wrench_provider returns all zero if it is not attached to the HumanStateProvider, without returning any error. We should either make it work even without a HumanStateProvider attached, or return an error in case it is not present.

cc @RiccardoGrieco

RiccardoGrieco, 2022-09-26:
I will open a PR to address this issue. For the time being, #317 should be a good workaround.

lrapetti, 2023-08-31:
With https://github.com/robotology/human-dynamics-estimation/pull/347, an error will be returned if no `IHumanState` interface is provided. We can wait for the PR and close it.

lrapetti, 2023-09-06:
Closed vua https://github.com/robotology/human-dynamics-estimation/pull/347

closed 2023-09-06 by lrapetti