gbionics/human-dynamics-estimation #337: Make HumanLogger compatible with `robot-log-visualizer`

lrapetti, 2023-02-13:
Making the `HumanLogger` data compatible with https://github.com/ami-iit/robot-log-visualizer would enable a quick analysis of the data.

In order to do so, we should simply modify the naming of some of the saved variables.

cc @GiulioRomualdi @RiccardoGrieco

S-Dafarra, 2023-02-13:
I guess that the main difference is how the joint values are parsed, right?

lrapetti, 2023-02-13:
I did a first preliminary investigation managing to modify an old human dataset in order to open it in the logger:

https://user-images.githubusercontent.com/35487806/218443056-2b7d277a-9f30-4eb6-ba1f-88a4ab617884.mov

The main changes I had to do are the following:
- change the data-struct name from `human_data` into `robot_logger_device`
- move joint configuration from `human_data.human_state.base_position` into `joints_state.positions`
- Added `description_list` with the list of joint names (in the original file this was empty)
- Added `yarp_robot_name` that was empty in the original file

(note that in order to properly save the modified from matlab, such that I was able to open it in the logger, I had to follow: https://github.com/h5py/h5py/issues/757#issuecomment-823757614)

Form the video we can observe there are still two major issue:
- time seems to be slower then expected
- other data are not properly loaded (we see #subsistem# in the data list)

S-Dafarra, 2023-02-13:
> * change the data-struct name from `human_data` into `robot_logger_device`
> * move joint configuration from `human_data.human_state.base_position` into `joints_state.positions`

It would be interesting to have the visualizer flexible on this. Ideally, we should be able to specify in some way which data to consider to visualize the robot

> * Added `description_list` with the list of joint names (in the original file this was empty)
> * Added `yarp_robot_name` that was empty in the original file

Similarly, we could ask the user to insert the robot name if missing, and assume a specific joint list.

> Form the video we can observe there are still two major issue:
> 
> * time seems to be slower then expected
> * other data are not properly loaded (we see #subsistem# in the data list)

Maybe it is worth checking with a robot log file and see if the same issue occurs

lrapetti, 2023-02-13:
> It would be interesting to have the visualizer flexible on this. Ideally, we should be able to specify in some way which data to consider to visualize the robot
>
>
> Similarly, we could ask the user to [...]

Concerning those points, if you think might be a good add-on, in the long term we can try to address those additions @GiulioRomualdi @S-Dafarra 

> > Form the video we can observe there are still two major issue:
> > 
> > * time seems to be slower then expected
> > * other data are not properly loaded (we see #subsistem# in the data list)
> 
> Maybe it is worth checking with a robot log file and see if the same issue occurs

I had no problem in opening a regular robot dataset (even when the human model was selected)

lrapetti, 2023-10-27:
> I opened the branch [`update_HumanLogger`](https://github.com/robotology/human-dynamics-estimation/tree/update_HumanLogger) to modify the logger; what I did was to follow what @lrapetti did here:
> 
> > The main changes I had to do are the following:
> > ```
> > * change the data-struct name from `human_data` into `robot_logger_device`
> > 
> > * move joint configuration from `human_data.human_state.base_position` into `joints_state.positions`
> > 
> > * Added `description_list` with the list of joint names (in the original file this was empty)
> > 
> > * Added `yarp_robot_name` that was empty in the original file
> > ```
> 
> Now the file logged can be directly uploaded in the `robot-log-visualizer` whiteout doing any modifications. Here is an example of a dataset that I logged:
> 
> [robot_logger_device_1696949792.014883.zip](https://github.com/robotology/human-dynamics-estimation/files/12859150/robot_logger_device_1696949792.014883.zip)

Those changes are quite relevant for all the people that are currently processing datasets acquired with `human-dynamics-estimation` software (e.g. @claudia-lat @LorenzoFiori @AntonellaTatarelli), and that then will require to change their processing software accordingly. Please let us know if you have something against those modifications, in particular:
- **change the data-struct name from `human_data` into `robot_logger_device`**
- **move joint configuration from `human_data.human_state.base_position` into `joints_state.positions`**

My idea is to eventually have a major release once this feature is merged (see https://github.com/robotology/human-dynamics-estimation/issues/373).

cc @evalli-iit @traversaro @Gianlucamilani @dariosortino @GiulioRomualdi

traversaro, 2023-10-27:
Probably at this point we may want to document this "protocol" (i.e. the name of variables in the .mat files) as now is not implemented only in blf/log-visualizer.

lrapetti, 2023-10-30:
> Probably at this point we may want to document this "protocol" (i.e. the name of variables in the .mat files) as now is not implemented only in blf/log-visualizer.

@traversaro do you mean documenting it within the context of this repo, or in a general entry-point?

traversaro, 2023-10-30:
> > Probably at this point we may want to document this "protocol" (i.e. the name of variables in the .mat files) as now is not implemented only in blf/log-visualizer.
> 
> @traversaro do you mean documenting it within the context of this repo, or in a general entry-point?

I mean somewhere, probably as at the moment we have many "writer"  and only one " reader", the reader is the winner, so the best place were to write this may be https://github.com/ami-iit/robot-log-visualizer .

lrapetti, 2023-10-30:
> Those changes are quite relevant for all the people that are currently processing datasets acquired with `human-dynamics-estimation` software (e.g. @claudia-lat @LorenzoFiori @AntonellaTatarelli), [...]

Ping to @claudia-lat @LorenzoFiori @AntonellaTatarelli @evalli-iit @dariosortino @Gianlucamilani in case you have anything against it, otherwise we can proceed merging this week.

PR #372 (2023-11-09): modified HumanLogger to be compatible with robot-log-visualizer
This PR closes #337; with this PR the matlab file saved by the `HumanLogger` can be opened by the `robot-log-visualizer`.
closed 2023-11-09 by lrapetti