gbionics/human-dynamics-estimation #343: Remove teleoperation models for icub2.5 and icub3

lrapetti, 2023-03-13:
The objective is to get rid of the teleoperation models found in https://github.com/robotology/human-dynamics-estimation/tree/master/conf/urdfs which are copies of the original robot urdf files (contained in https://github.com/robotology/icub-models) with the addition of fake links originally added to replicate the orientation of the sensors data.

Those links added to the urdfs are no longer necessary after https://github.com/robotology/human-dynamics-estimation/pull/254.

In fact, for example, the `ergoCub` model was added and tested (https://github.com/robotology/human-dynamics-estimation/pull/340) by using directly the original model by adding all the required fixed transforms to the configuration file (see https://github.com/robotology/human-dynamics-estimation/blob/11ad67f1d31f54d42aa1954a6a1a6e83a5892570/conf/xml/RobotStateProvider_ergoCub_openxr_ifeel.xml#L104-L137). Since the `ergoCub` model should basically have same (or similar) link definition as the `iCub3` (and `iCub2.5`) models, starting from those transforms we should be able to get rid of the teleoperation models.

cc @mebbaid @S-Dafarra

S-Dafarra, 2023-03-13:
> Since the `ergoCub` model should basically have same (or similar) link definition as the `iCub3` (and `iCub2.5`) models, starting from those transforms we should be able to get rid of the teleoperation models.

I remember that with @GiulioRomualdi we had some issues in understanding the frame wrt which the offset was defined (the rotation part was clearer). When using the offsets of iCub3 (https://github.com/robotology/human-dynamics-estimation/blob/11ad67f1d31f54d42aa1954a6a1a6e83a5892570/conf/xml/RobotStateProvider_iCub3_openxr_ifeel.xml#L111-L118) we did not have the expected effect.

lrapetti, 2023-03-27:
I noticed that we are also maintaining some files for whole-body retargeting (e.g. https://github.com/robotology/human-dynamics-estimation/blob/master/conf/xml/RobotStateProvider_iCub2_5.xml) so in order to get rid of the teleoperation model we should also add the fixed transform for the leg frames (so it's not simply about copying ergoCub configuration files). 

One option might be to remove those configuration for whole-body retargeting since they are currently not being used.