gbionics/human-dynamics-estimation #346: Remove implementation of both IWrapper and IMultipleWrapper from devices

RiccardoGrieco, 2023-03-22:
As per title, this issue is to replicate what we did in #344 as a solution for #342.

Currently the devices listed below implement both the interfaces `IWrapper` and `IMultipleWrapper`, which makes them prone to bugs (see above).

Affected devices:
- [ ] [HumanControlBoard](https://github.com/robotology/human-dynamics-estimation/tree/master/devices/HumanControlBoard)
- [ ] [HumanDynamicsEstimator](https://github.com/robotology/human-dynamics-estimation/tree/master/devices/HumanDynamicsEstimator)
- [ ] [HumanStateProvider](https://github.com/robotology/human-dynamics-estimation/tree/master/devices/HumanStateProvider)
- [ ] [HumanWrenchProvider](https://github.com/robotology/human-dynamics-estimation/tree/master/devices/HumanWrenchProvider)
- [ ] [RobotPositionController](https://github.com/robotology/human-dynamics-estimation/tree/master/devices/RobotPositionController)

cc @lrapetti

RiccardoGrieco, 2023-03-24:
Changes were implemented in branch https://github.com/robotology/human-dynamics-estimation/tree/devices-fix_wrapper_interfaces, opened draft PR #347 .

lrapetti, 2023-09-06:
Closed via https://github.com/robotology/human-dynamics-estimation/pull/347

closed 2023-09-06 by lrapetti