gbionics/human-dynamics-estimation #358: Replace usage of `analogServer`

lrapetti, 2023-09-28:
Currently we are using the `analogServer` in order to stream the wrench data from `IHumanWrench` interface (see https://github.com/search?q=repo%3Arobotology%2Fhuman-dynamics-estimation%20analogServer&type=code). However, the `analogServer` corresponds to the `analogWrapper` mentioned in https://github.com/robotology/yarp/pull/2999, which will be soon be deprecated.

Indeed we should understand what are the alternatives. One option could be to implement our own message, and wrapper/remapper devices as we are doing for the `IHumanState` and `IHumanDynamics`, and removing the `IAnalogSensor` interface from the wrench provider (https://github.com/robotology/human-dynamics-estimation/blob/550c5603d549be4a648301eb57b391631daedca1/devices/HumanWrenchProvider/HumanWrenchProvider.h#L49-L56). Clearly this will also break the `ROS` pipeline, which anyway should be soon addressed (see also https://github.com/robotology/human-dynamics-estimation/issues/335).

Another option, would be to understand what is the alternative/equivalent device

lrapetti, 2023-09-28:
cc @traversaro

traversaro, 2023-09-28:
For the ROS part, you can use `WrenchStampedRosPublisher` as long as you expose the data also with a [`yarp::dev::ISixAxisForceTorqueSensors`](https://www.yarp.it/git-master/classyarp_1_1dev_1_1ISixAxisForceTorqueSensors.html). For the YARP part, who is using `/HDE/HumanWrenchPublisher/wrench:o` ? There is any assumption that the data is in vector form?

lrapetti, 2023-09-28:
I have opened https://github.com/robotology/human-dynamics-estimation/issues/361 to address the replacement of `analogServer` for what concern the YARP messages. It will remain the `analogServer` to stream data in ROS which should be replaced.

claudia-lat, 2024-05-10:
I think we can close this issue, am I right?

S-Dafarra, 2024-05-10:
> Apparently three files are still using that:
> 
> * https://github.com/search?q=repo%3Arobotology%2Fhuman-dynamics-estimation%20analogServer%20&type=code
> 
> I would remove the files before closing the issue. If instead the files are still used, then we have a problem. fyi @S-Dafarra @mebbaid

I think this is similar to what @dariosortino did for the iFeel part. Those wrappers were useful to stream over ROS the values of the shoes I suppose. I doubt they work anyhow, since the ROS group has disappeared from recent versions of the device.

mebbaid, 2024-05-10:
As far as I know
- on the `human.xml` side, the main users are the iFeel human wrenches visualization guys. I think they already moved from streaming the data in that way @dariosortino may confirm.
- The PHRI XML files were used by @lrapetti for his experiments. Those same experiments will be reproduced by @CarlottaSartore and in this case it maybe a good chance to remove them and adapt to new versions of the device

dariosortino, 2024-05-13:
> * on the `human.xml` side, the main users are the iFeel human wrenches visualization guys. I think they already moved from streaming the data in that way @dariosortino may confirm.

Sorry but about this I don't really know much what's going on currently

traversaro, 2024-05-13:
> > * on the `human.xml` side, the main users are the iFeel human wrenches visualization guys. I think they already moved from streaming the data in that way @dariosortino may confirm.
> 
> Sorry but about this I don't really know much what's going on currently

I guess it all boils down to which files you are using in the iFeel-Human demos. If you are not using https://github.com/robotology/human-dynamics-estimation/blob/48d0147a69572cb2d3c3cf06c421d2703e0bc2ed/conf/xml/Human.xml#L197 anymore, then I guess we can remove them. fyi @claudia-lat @GiulioRomualdi

dariosortino, 2024-05-13:
Checking in our [configuration files](https://github.com/ami-iit/ifeel_utils/tree/main/software/iFeelHDE/conf), it seems that we don't use that anymore

traversaro, 2024-05-13:
Perfect, I guess we can delete the `human.xml` in this repo then.

lrapetti, 2024-05-13:
> * The PHRI XML files were used by @lrapetti for his experiments. Those same experiments will be reproduced by @CarlottaSartore and in this case it maybe a good chance to remove them and adapt to new versions of the device

Those file are actually older, I think they were used for An.Dy. demo before I even joined! Concerning `Human.xml` that's the file that was used for Xsens, I don't know if it might still be needed.

dariosortino, 2024-05-13:
It would be nice also to correct this https://github.com/robotology/human-dynamics-estimation/blob/master/misc/hde_scheme.png

traversaro, 2024-05-13:
> It would be nice also to correct this https://github.com/robotology/human-dynamics-estimation/blob/master/misc/hde_scheme.png

@lrapetti do you happen to know where the source of that drawing is?

claudia-lat, 2024-05-13:
@traversaro I've got the .svg of the scheme, I will fix it together with @davidegorbani.

mebbaid, 2024-05-13:
you can use the updated figure discussed in the handover in https://github.com/ami-iit/element_ergonomy-control/issues/183#issuecomment-1872496045