gbionics/human-dynamics-estimation #366: HumanStateVisualizer is stuck waiting data from HumanStateWrench

davidegorbani, 2023-10-10:
I tried to launch the `HumanStateVisualizer` on a terminal and it was stuck on waiting data from the `HumanStateWrench` device; the output on the terminal was:
```
[WARNING] HumanStateVisualizer : 'visualizeFrames' option not found or not valid. It is set to False.
[DEBUG] |yarp.dev.PolyDriver|human_state_nwc_yarp| Parameters are (device human_state_nwc_yarp) (humanStateDataPort "/HDE/HumanStateServer/state:o")
[DEBUG] HumanState_nwc_yarp : Configuring input data ports
[INFO] |yarp.os.Port|/tmp/port/2| Port /tmp/port/2 active at tcp://10.240.9.164:10050/
[DEBUG] HumanState_nwc_yarp : Opening input ports
[INFO] |yarp.os.impl.PortCoreInputUnit|/tmp/port/2| Receiving input from /HDE/HumanStateServer/state:o to /tmp/port/2 using tcp
[DEBUG] HumanState_nwc_yarp : Opened correctly
[INFO] |yarp.dev.PolyDriver|human_state_nwc_yarp| Created device <human_state_nwc_yarp>. See C++ class hde::devices::HumanState_nwc_yarp for documentation.
[INFO] HumanStateVisualizer : Waiting for data from HumanStateClient
[INFO] HumanStateVisualizer : Human State Interface providing data for the base link [  Pelvis  ]
[INFO] HumanStateVisualizer : Human State Interface providing data from [  31  ] joints
[DEBUG] |yarp.dev.PolyDriver|human_wrench_nwc_yarp| Parameters are (device human_wrench_nwc_yarp) (humanWrenchDataPort "/HDE/HumanWrenchServer/wrench:o")
[DEBUG] HumanWrench_nwc_yarp : Configuring input data ports
[INFO] |yarp.os.Port|/tmp/port/3| Port /tmp/port/3 active at tcp://10.240.9.164:10051/
[DEBUG] HumanWrench_nwc_yarp : Opening input ports
[INFO] |yarp.os.impl.PortCoreInputUnit|/tmp/port/3| Receiving input from /HDE/HumanWrenchServer/wrench:o to /tmp/port/3 using tcp
[DEBUG] HumanWrench_nwc_yarp : Opened correctly
[INFO] |yarp.dev.PolyDriver|human_wrench_nwc_yarp| Created device <human_wrench_nwc_yarp>. See C++ class hde::devices::HumanWrench_nwc_yarp for documentation.
[INFO] HumanStateVisualizer : Waiting for data from HumanWrenchClient
[INFO] HumanStateVisualizer : Waiting for data from HumanWrenchClient
[INFO] HumanStateVisualizer : Waiting for data from HumanWrenchClient
[INFO] HumanStateVisualizer : Waiting for data from HumanWrenchClient
```

I tried to read from the YARP port `/HDE/HumanStateServer/state:o` and it was streaming data.

lrapetti, 2023-10-17:
Hi @davidegorbani, the problem is due to the fact that starting from https://github.com/robotology/human-dynamics-estimation/pull/362/commits/01812b072279543b369415d0e585a89b021193c6 the `HumanStateVisualizer` uses the `HumanWrench` interface. Indeed, the `Human.xml` file you are using should have the same change as https://github.com/robotology/human-dynamics-estimation/pull/362/commits/01812b072279543b369415d0e585a89b021193c6#diff-90882fd47b6c2e86b383770e476062185beb69ba0b5dceacdfb8b8280b384819R185-R193 to expose the wrench data.

lrapetti, 2023-10-19:
> Hi @davidegorbani, the problem is due to the fact that starting from [01812b0](https://github.com/robotology/human-dynamics-estimation/commit/01812b072279543b369415d0e585a89b021193c6) the `HumanStateVisualizer` uses the `HumanWrench` interface. Indeed, the `Human.xml` file you are using should have the same change as [01812b0#diff-90882fd47b6c2e86b383770e476062185beb69ba0b5dceacdfb8b8280b384819R185-R193](https://github.com/robotology/human-dynamics-estimation/commit/01812b072279543b369415d0e585a89b021193c6#diff-90882fd47b6c2e86b383770e476062185beb69ba0b5dceacdfb8b8280b384819R185-R193) to expose the wrench data.

This fixed the issue (see https://github.com/robotology/human-dynamics-estimation/pull/368#issuecomment-1766019700). I'll close this issue.

closed 2023-10-19 by lrapetti