gbionics/human-dynamics-estimation #369: CI failing on Windows

lrapetti, 2023-10-18:
See failure at https://github.com/robotology/human-dynamics-estimation/actions/runs/6544107072/job/17769907998

lrapetti, 2023-10-18:
@davidegorbani @traversaro any suggestion?

traversaro, 2023-10-18:
Those are conflicts between the conda installed by setup-conda and the one installed inside GitHub Actions. The possible fixes are: 
* Simple: force the use of mambaforge, as done in https://github.com/robotology/robometry/commit/be84ef5d9696cfb9909f45ca15eeea534944c93d
* More complex, but resulting in a faster CI: switch to setup-micromamba, as done in https://github.com/robotology/yarp-device-xsensmt/pull/46/commits/9062fc3ed7cf0bed4d3c20507c95e691a27f03f5

lrapetti, 2023-10-19:
Thank @traversaro 

> * More complex, but resulting in a faster CI: switch to setup-micromamba, as done in [robotology/yarp-device-xsensmt@9062fc3](https://github.com/robotology/yarp-device-xsensmt/commit/9062fc3ed7cf0bed4d3c20507c95e691a27f03f5)

I have implemented this solution (see https://github.com/robotology/human-dynamics-estimation/pull/370), we can close it.

closed 2023-10-19 by lrapetti