gbionics/human-dynamics-estimation #373: Release human-dynamics-estimation v3.0.0

lrapetti, 2023-10-27:
Since https://github.com/robotology/human-dynamics-estimation/pull/372 and https://github.com/robotology/human-dynamics-estimation/pull/367 will break compatibility with previous configurations and dataset, I think we can consider releasing a `v3.0.0` version of the `human-dynamics-estimation` software. What do you think @traversaro? Should we consider instead maintaining the deprecated devices in https://github.com/robotology/human-dynamics-estimation/pull/367?

traversaro, 2023-10-27:
I agree with a clean new major and breaking compatibility, however @S-Dafarra and @mebbaid may have relevant opinions.

mebbaid, 2023-10-27:
I agree with a new release. I will tag  @claudia-lat @AntonellaTatarelli @LorenzoFiori in this since they are currently using `hde` for various activities and may prefer to use  `HumanLogger` with original naming convention for the short term before the breaking changes in #372

PR #375 (2023-11-14): Bump human-dynamics-estimation version to 3.0.0
Closes #373.
Before merging:
- [x] Update documentation with the mapping of the new device names @davidegorbani
- [x] Replace usage of `transformClient` in the configuration files (https://github.com/robotology/wearables/issues/191) @S-Dafarra 
- [x] Replace usage of `controlboardwrapper2` in the configuration files @lrapetti  https://github.com/robotology/human-dynamics-estimation/pull/378

### Changed
- Naming convention from `wrapper`\ `server` to `nws`\ `nwc` (https://github.com/robotology/human-dynamics-estimation/pull/367).
- HumanLogger de [... cut]
closed 2023-11-14 by lrapetti